MarlinFirmware/Marlin #17453: [BUG] Inconsistent Action Command reporting during filament runout

patriot1889, 2020-04-09:
### Bug Description

With "HOST_ACTION_COMMANDS" enabled, when a filament runout is detected it sends the following to the host;
`//action:paused filament_runout 0`
The number indicates which tool has runout.

However, Marlin will only notify the host on the first filament runout, subsequent runouts are not declared.

### My Configurations

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/4457726/Archive.zip)

### Steps to Reproduce

1. Start a print
2. Cause filament runout by cutting filament
3. Replace filament
4. Resume print
5. Cause second filament runout by cutting filament.

**Expected behavior:** 

Each filament runout should send the correct HOST_ACTION_COMMAND

**Actual behavior:** 

Only the first filament runout is declared through a HOST_ACTION_COMMAND

#### Additional Information
Looking back at the serial response, there is no "resumed" action returned to the host once the print is resumed after the first runout. Could this be a reason? Does the HOST_ACTION_COMMANDS logic think the printer is still paused?

boelle, 2020-06-20:
@patriot1889 is the issue still there?

patriot1889, 2020-06-20:
> @patriot1889 is the issue still there?

I shall check and report back.

boelle, 2020-06-29:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-29 by boelle