MarlinFirmware/Marlin #17455: BTT SKR V1.4 Turbo FAN_PIN error P2_03 Not good.

AntoszHUN, 2020-04-09:
Hi! In Marlin version 2.0.3, code P2_03 does not work next to E0_AUTO_FAN_PIN. P2_04 is working. But P2_03 is not. Motherboard is a BigTreeTech SKR V1.4 Turbo. I want to use one as an object cooler. The other cools the hotend, which is 60 ° C on / off.

thisiskeithb, 2020-04-09:
FAN0/`P2_03` is for the parts cooling fan and HE1/`P2_04` is for the hotend fan. You'll have to modify [`pins_BTT_SKR_common.h`](https://github.com/MarlinFirmware/Marlin/blob/5158106c12faf961eb886cbc6348060ce093c340/Marlin/src/pins/lpc1768/pins_BTT_SKR_common.h#L70-L90) if you want to remap them.

Also, you can simply use `#define E0_AUTO_FAN_PIN  FAN1_PIN` instead of referencing the pin number directly.

AntoszHUN, 2020-04-10:
Thanks for the reply! I don't have a pins_BTT_SKR_common.h file in Marlin. Maybe the pins_BTT_SKR.h file would be smooth? What do I need to change?

thisiskeithb, 2020-04-10:
Make sure you're running the latest version ([`2.0.5.3`](https://github.com/MarlinFirmware/Marlin/releases/tag/2.0.5.3) as of this post) and start from there.

AntoszHUN, 2020-04-10:
It's okay.

AntoszHUN, 2020-04-11:
I also did it with version 2.0.5.3. The error is still the same. Nothing has changed. Then let's make a mistake. What do I need to do to be good? Can you help?
[image]

thisiskeithb, 2020-04-11:
Did you remap the pins? Like I mentioned above:

> FAN0/`P2_03` is for the parts cooling fan and HE1/`P2_04` is for the hotend fan. You'll have to modify [`pins_BTT_SKR_common.h`](https://github.com/MarlinFirmware/Marlin/blob/5158106c12faf961eb886cbc6348060ce093c340/Marlin/src/pins/lpc1768/pins_BTT_SKR_common.h#L70-L90) if you want to remap them.

Also, this is not a bug, so please close the issue.

ellensp, 2020-04-11:
It is very clear, you **cannot** I repeat, **cannot** use the same pin for FAN_PIN and E0_AUTO_FAN_PIN
FAN_PIN is the plastic parts cooling fan, it is controlled by the g-code produced by the Slicer.  
E0_AUTO_FAN_PIN is the hot end cooling fins fan. It is automatically turned on when the hotend is warmer that 50c.
Not a bug, but a PEBKAC issue.

AntoszHUN, 2020-04-11:
Where is FAN0 here? I can not find it. There is only FAN but there is pin P2_03 there at the factory.
[image]

AntoszHUN, 2020-04-11:
I want to use one as an object cooler. The other would be used to cool the hotend. Because E0_AUTO_FAN_PIN is the heatend cooler that is plugged into P2_03 for me. I want to use this as a hotend cooler. The other HE1 will be the object cooler. I will close the discussion if I understand the logic of the whole setup. There is no problem here. I apologize for the inconvenience.

superkefir, 2020-04-11:
You should remap ping at line
#define FAN_PIN                          P2_03
to #define FAN_PIN                          P2_04
Then you can use line #define E0_AUTO_FAN_PIN P2_04

closed 2020-04-11 by AntoszHUN