MarlinFirmware/Marlin #17456: [BUG] Ramps Heated chamber compiles but doesn't work. Missing TEMP_CHAMBER_PIN.

KevMags, 2020-04-09:
### Bug Description

Any version in the last year of Marlin with ramps 1.4.  Heated chamber not enabled.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/4458409/Configuration.h.zip)
[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/4458412/Configuration.h.zip)

### Steps to Reproduce

1. [First Step]. Enable Heated chamber by defining #define TEMP_SENSOR_CHAMBER 1. #define HEATER_CHAMBER_PIN       44.  Upload to Ramps
2. [Second Step]No items in LCD menu referencing Heated Chamber. Send M141 S30, Terminal returns "Unknown Command".
3. [and so on...]

**Expected behavior:** [What you expect to happen]
Heated chamber items in LCD and M141 gcode working. 

**Actual behavior:** [What actually happens]

No items in LCD menu referencing Heated Chamber. Send M141 S30, Terminal returns "Unknown Command"

#### Additional Information
Added "TEMP_CHAMBER_PIN 15" (E2 temp sensor) to the "pins_RAMPS.h" (Line 171) file.
Everything works fine now.
[Pins file screenshot.png.zip](https://github.com/MarlinFirmware/Marlin/files/4458487/Pins.file.screenshot.png.zip)

[pins_RAMPS.h.zip](https://github.com/MarlinFirmware/Marlin/files/4458476/pins_RAMPS.h.zip)

 

Fixed my editing 

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

boelle, 2020-06-20:
@KevMags is the issue still there?

KevMags, 2020-06-20:
Yes

boelle, 2020-06-23:
Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

sjasonsmith, 2020-07-04:
@KevMags, is the problem just that nothing warned you during compilation that you were missing that pin? It sounds like everything is working once you added the missing custom pin?

It looks like a Sanity Check was added to the `bugfix-2.0.x` branch 15 days ago, which should do just that.
d8c3aed0383202341a29e65fa556a55c5af47185

If this is not what you had in mind, please describe what change you are expecting from this issue.

KevMags, 2020-07-04:
Everything worked correctly after adding the pin assignment.  I'll check the new bugfix.  Thanks

KevMags, 2020-07-04:
Nailed it.  Thank you for the help.

sjasonsmith, 2020-07-06:
@KevMags based on that last comment I assume the sanity check is all you needed here. I'll close this issue, please reply or re-open it if you think something was missed.

closed 2020-07-06 by sjasonsmith