MarlinFirmware/Marlin #17457: SKR V1.4 TURBO WITH TMC2209 drivers, extruder issue

whittouk, 2020-04-09:
### Bug Description

Recently changed my standard creality ender 3 pro board out for a SKR v1.4 Turbo with TMC2209 drivers. Managed to config it all but obviously not correctly as the extruder will not turn when printing. All X,Y,Z axis work correctly to my knowledge. Im totally new to this and now thinking ive bitten off more i can chew! Hopefully someone will be kind enough to help me!

The printer is only 2 days old but read reviews i could quieten it with the above so went to task on it however have hit this issue. so i know the stepper is ok and i have swapped cables to check cabling with other steppers and everything moves and operates so i know it must be down to the configuration.

### My Configurations

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/4458801/Configuration_adv.zip)

**Expected behavior:** Guidance on getting the extruder to work during print

**Actual behavior:** Nothing

#### Additional Information

KevMags, 2020-04-09:
I the nozzle heating up? The nozzle needs to be hot before it will extrude.

whittouk, 2020-04-09:
yes the nozzle gets hot 205 i have it set at and bed at 50

ellensp, 2020-04-10:
Adjust your e0 stepper current. use GCODE M906 E<mA>  where mA is from 0 to 2000. your configuration says your at 900.  I would expect you need at least 1200 
Once you have found where it works update the firmware "#define E0_CURRENT      900" with new value. and just to be sure save to eeprom with M500

whittouk, 2020-04-10:
hi 
again nothing happens.it doesnt turn :-(

[image]

ellensp, 2020-04-10:
What does M122 say? 
Are you sure you plugged the E0 stepper into E0M plug and not ZMB plug?

whittouk, 2020-04-10:
Below is results

[image]

whittouk, 2020-04-10:
i think i may have just found it ! after re-reading your comment on which its plugged into i think i had it in the wrong place on the board it was in ZMB!

whittouk, 2020-04-10:
BINGO!!! THANK YOU Ellen!!!

ellensp, 2020-04-10:
Since its solved please close this issue.

whittouk, 2020-04-10:
is there anyway of backing up my configuration file just incase?

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle