MarlinFirmware/Marlin #17463: [BUG] (Can't get Marlin to work with MKS SGEN L)

TheMath38, 2020-04-10:
#16838 # Bug Description

Hey ! I recently got my MKS SGEN L with some TMC2209 stepper motors. I've built and compiled in VScode (which result successful) the firmware, customize it etc. Then, I've uploaded the firmware.bin file onto the root of the SD CARD (called REARM) and... Nothing happend. Whatever i'm doing, I can't connect my mainboard in my computer, can't get my thermistors temperature level, can't move any stepper motors... 

I switched to Smoothieware to see if it was a motherboard issue, but it worked half (The motor were moving only in one side etc)
### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
Here's the Marlin folder: [Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4460739/Marlin.zip)

### Steps to Reproduce

1. [First Step] : Compile and build in VS code my configuration file (check in PlatformIo for the correct environment, such as LPC1768)
2. [Second Step] Upload it on the MKS SGEN L board via SD.
3. [and so on...] ANd... Nothing happen, can't connect to USb and can't get any information etc etc

**Expected behavior:** [What you expect to happen] The printer to work xD

**Actual behavior:** [What actually happens]
Litteraly nothing. I can't do ANYTHING
#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
here is a link for a video which will explain the problem better than me: https://mega.nz/file/x1Z3HKbT#3cdzWoH9wKBSeqFOEWqCyKjVAVeR1VJDKg_9dVnjHZk
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-04-10:
You have #define SERIAL_PORT -1 which should enable serial over usb as primary controller
but you also have //#define SERIAL_PORT_2 -1 
The second serial should be configured as usb and the primary serial should be set to your touch screen. 
Ie It should be 
#define SERIAL_PORT 0
#define SERIAL_PORT_2 -1
  
And for goodness sake turn your darn phone around 90 degrees so the video is in Landscape mode.  Portrait mode is useless for this.

ellensp, 2020-04-10:
The unknown usb device is worrying... 3rd one I've seen today... can't seem to find a cause.  Compiles and works for me. Try this. Your configs + serial changes on latest marlin bugfix-2.0.x compiled on my machine.  In a zip file. unzip before use.     
[firmware.zip](https://github.com/MarlinFirmware/Marlin/files/4461304/firmware.zip)

TheMath38, 2020-04-10:
Hey, thanks for the fast reply ^^
I changed the settings like you said, cleaned up, recompiled, transfered it and... Same issue. 
I don't really know what is really happening inside the mainboard, it's going to make me crazy arrrg x__x

Ho, and srory for the video x) I didn't think about turning my phone :3

TheMath38, 2020-04-10:
Okay, I installed your firmware, and so on. Can't do anything on both my touchscreen and my computer (which continues to not recognize my mainboard huu)

hamster65, 2020-04-10:
I dont't have a touch screen. I compiled and flashed the bugfix branch on my SGEN L and it works without a problem with these settings:
#define SERIAL_PORT -1
//#define SERIAL_PORT_2 -1
#define BAUDRATE 250000

Is your board reading the SD card correctly? Maybe it is the wrong file system and needs to be re-formated. I think the LEDs on the board can be used for debugging.

TheMath38, 2020-04-10:
Hmm, I changed my settings like you. My computer don't recognize my motherboard...
Ho, and I didn't said that before I switched to Smoothieware, I installed the driver and guess what? It worked! I was able to see my motherboard (COM5).

But when I switched to Marlin, nothing. I just have the D6 LED alight, shining in red, and some seconds later, the D1 LED is doing the same, but in blue.

hamster65, 2020-04-10:
I have the red LED (labeled D8 on my board!) and a blue one (D1) next to it. The blue one is on, but turns of every second.

TheMath38, 2020-04-11:
Okay, so I checked a loooot of forums, including this one, and I figured a similar issue like me (but instead of using marlin, he was using grbl-LPC). Here is the link : https://forum.makerforums.info/t/mks-sgen-cant-connect-to-motherboard-with-grbl-lpc/77896

i figured out that the problem seems to be an I2C chip that is not present on my mainboard. So when I boot up the mainboard, it is crashing due to the non presence of it. 

So in order to fix the problem by my own, I searched on the marlin FW settings, without any proud results sadly...
Maybe I missed something? If you could investigate on it I would very appreciate it ^^

TheMath38, 2020-04-13:
Please... Could somebody help me to fix this issue? I'm very lost...

By the way, I tried 3 different SD card, and it still doesn't work... :3

ellensp, 2020-04-13:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted [on this page.](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question)) For best results getting help with configuration and troubleshooting, please use the following resources:

MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: [https://discord.gg/n5NJ59y](https://discord.gg/n5NJ59y)
After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

Kyosuke33, 2020-05-22:
I have exactly the same problem with my Mks Sgen L.
Can't do anything after boot with Marlin Firmware on the sd card.
When i connect the board to the computer, i can't do anything, no com port, i just can see SMOOTHIE with a yellow triangle on the hardware. (i never install smoothieware but maybe it's install by default.

TheMath38, 2020-05-22:
> I have exactly the same problem with my Mks Sgen L.
> Can't do anything after boot with Marlin Firmware on the sd card.
> When i connect the board to the computer, i can't do anything, no com port, i just can see SMOOTHIE with a yellow triangle on the hardware. (i never install smoothieware but maybe it's install by default.
> 
> You find a solution ? Thank you.

I've found the solution, the issue was that my thermistor wasn't configured correctly. So when the firmware was booting, Marlin was reading 90000°C and directly went to a thermal runaway error.
To fix that I just disabled the thermal runaway protection in the configuration_adv.h file and this fixed the issue!

(My thermistor was broken so I've ordered one of thoses PT100 on Aliexpress, I fairly recommend you the TriangleLab's sensor)
Hope this will help you :)

Kyosuke33, 2020-05-22:
Ok i try this thank you !

TheMath38, 2020-05-22:
> Ok i try this thank you !

Ho problem :) Hope this will help you

Kyosuke33, 2020-05-23:
Just try, not the thermistor BUT ! knowing that i try by remove all the component one buy one, and it was the driver LV8729 for the extruder who make the problem. Change buy an other it work !

Thank you for your help !

closed 2020-06-21 by boelle