MarlinFirmware/Marlin #17464: Marlin 2.0.x G29 with BLTouch dont move the Z down when probing

Flakigel, 2020-04-10:
Hallo,

I only want to use the BLTouch only for AUTO_BED_LEVELING_BILINEAR. My printer is an Anycubic Mega S with dualenstops switch on Z. They use #define USE_ZMIN_PLUG and
#define USE_XMAX_PLUG.

G28 will work fine. The probe didn´t use for it. 
When I start after G29, the probe drive to first grid point an deploy 3 times fast in the air and get an error probing. The Z Axis dont move down to the bed. 

Is this BUG? With marlin 1.1.9 its works great.

When I using the Probe for homeing, thats works great. 
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4461537/Marlin.zip)

kolbenhans, 2020-04-10:
Try this https://github.com/MarlinFirmware/Marlin/issues/17440#issuecomment-611727891

Flakigel, 2020-04-13:
Thanks. That’s help a little bit. Z moves down to probe. But when I try tonest the z offset for probe, it didn’t work with the offset set to 0 before. I will probe one times, then the BLTouch is blinking, the printer restart g28 I think. He moves all axis home. But z will then crash into the bed. Because the Endstop switch doesn’t work.

boelle, 2020-06-20:
@Flakigel is the issue still there?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle