MarlinFirmware/Marlin #17466: [BUG] Unable to build Marlin for SKR Pro

Luke-Elsdon, 2020-04-10:
### Bug Description
Unusual build errors when trying to build Marlin 2.0 bugfix for an SKR Pro. Have also tried non-bugfix, and bigtreetech's own build of Marlin. I have tried a multitude of fixes regarding changing libraries and using different combinations of Marlin/TMC Libraries etc but can't seem to get any to work. This is the first time I have configured firmware from scratch and is required due to making a non-standard printer, so it may be a configuration issue, however I cannot find anyone with the same issue and other general issues with this board have been attempted.

### My Configurations

 I have attached my config files and the error output.

hills8, 2020-04-10:
try to change:
#define SERIAL_PORT_2 6 // Second communication method (6 is WiFi via ESP-01S)
to:
#define SERIAL_PORT_2 1 // Second communication method (6 is WiFi via ESP-01S)

Luke-Elsdon, 2020-04-10:
Not sure why I set the baudrate to 250000, thanks. I've changed it now.
Unfortunately swapping those values round hasn't helped.
I'm convinced this is a dependency issue, I found the following config on another thread:
[platformio.txt](https://github.com/MarlinFirmware/Marlin/files/4462976/platformio.txt)
When I use this it seems to get much further, but then I get the following message right at the end:

`/home/luke/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: cannot open linker script file stm32f407xg.ld: No such file or directory
collect2: error: ld returned 1 exit status
*** [.pio/build/BIGTREE_SKR_PRO/firmware.elf] Error 1`

Bob-the-Kuhn, 2020-04-10:
I tried to reproduce your problem using today's bugfix-2.0.x and came up with different errors.

It looks like you are attempting to use old software for bugfix-2.0.x, platformio.ini, configuration.h and configuration_adv.h.  

I suggest starting with a fresh copy of bugfix-2.0.x and then manually copying your configuration.h and configuration_adv.h items over to the new files.

Also, I made the following changes to my pins_**BTT_SKR_PRO_V1_1.h** file:
- add: `#define E0_AUTO_FAN_PIN                     PC9`
- uncomment `#define FLASH_EEPROM_EMULATION`

Luke-Elsdon, 2020-04-10:
Thanks Bob, I've done what you said and I can finally build the firmware. No idea how I managed to use an older version, but I have been trying a ton of things to fix this so I've likely done it then.
Will close this now.

closed 2020-04-10 by Luke-Elsdon