MarlinFirmware/Marlin #17472: Cant compile dev-2.1.x with controller_fan set up

Exellent1988, 2020-04-10:
### Bug Description

Cant compile the dev version of 2.1.x of mRalin when i switch on controller_fan

### My Configurations
SKR 1.4
Relativly default but controller fan won't compile:

### Steps to Reproduce

enable controller_fan and define a pin

**Expected behavior:** compile fine

**Actual behavior:** compile won't complete

#### Additional Information
COMPILE ERROR:

In file included from Marlin\src\MarlinCore.cpp:173:
Marlin\src\feature/controllerfan.h:61:46: error: 'constexpr' needed for in-class initialization of static data member 'const controllerFan_settings_t& ControllerFan::settings' of non-integral type [-fpermissive]
   61 |       static const controllerFan_settings_t &settings = controllerFan_defaults;
      |                                              ^~~~~~~~
*** [.pio\build\LPC1768\src\src\MarlinCore.cpp.o] Error 1

ogsv, 2020-04-11:
Not a solution, but active development is currently on the bugfix-2.0.x branch.

TheDark, 2020-04-14:
This has already been fixed in the bugfix-2.0.x branch.

The solution is to replace `Marlin/src/feature/controllerfan.h:61´

```
      static const controllerFan_settings_t &settings = controllerFan_defaults;
```

with

```
      static const controllerFan_settings_t constexpr &settings = controllerFan_defaults;
```

closed 2020-04-16 by thinkyhead