MarlinFirmware/Marlin #17473: Stealthcop bad sound. latest bugfix-04/11

Hagbard-84, 2020-04-10:
Stealthcop stop working, random event, bad sound coming from E0, if I restart printer it's back too normal ore if i turn it off via lcd

### 

### My Configurations
Ender 3, cheetha 1.1

**Expected behavior:** silence from E0]

**Actual behavior:**  bad sound coming from E0]

#### Additional Information

https://www.dropbox.com/s/8yb42nfs5p62r83/E0.m4a?dl=0

thinkyhead, 2020-04-22:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle