MarlinFirmware/Marlin #17475: [BUG] MKS Base v1.6 does not auto home or recognize temperature sensors

krawson14, 2020-04-11:
### Bug Description

Installed on new MKS Base v1.6 motherboard and have seen the following issues with the install on my Ender 3.

1. LCD screen lights up but does not provide any sort of output
2. Autohome produces a command, but the Y motor buzzes, does not move the bed, and X moves about +5 mm. Trying to control the printer from octoprint will only work in the positive direction in X and Y axis, but does both positive and negative directions in Z. 
3. I have three temperature sensors, one for the bed, and one for each hot end that I installed. Only one of the hot ends is recognized, and when I power on the printer the unrecognized hot end begins to heat without any data being reported and without control. I have switched around the pin connectors for the sensors, but see no data being reported. 

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Flashed mother board with Marlin 2.0 through Arduino IDE
2. Powered on printer
3. Saw no output on the LCD pannel
4. Connected with Octoprint
5. Saw no temperature data being recorded
6. Noticed smoke coming from hot end #2 as it was heating without control
7. Powered off

**Expected behavior:** 
I would expect that auto home, or control in both positive/negative directions for X and Y axis to be available.
I would expect temperature data to be available to Octoprint as well as to the firmware.
I would expect the LCD screen to show some sort of output

**Actual behavior:**
Auto home does not actually move the hot end to home, and no control in negative direction for the X and Y axis
No temperature data available. Nothing shows besides hot end 1 in Octoprint. 
No output on LCD screen.

ellensp, 2020-04-11:
**Please include a ZIP file containing your Configuration.h and Configuration_adv.h files.**

krawson14, 2020-04-11:
[config_files.zip](https://github.com/MarlinFirmware/Marlin/files/4465139/config_files.zip)
Attached config files for Marlin.

krawson14, 2020-05-01:
After playing with the configs quite a bit, I was able to get the sensors to recognize the hotends and bed, and it no longer heads with endless abandon. 

However I am still having issues getting the home controls to work properly. I have figured out through trial and error how to set up the axis directions, however whenever I try to auto home, Z travels up and stops after about 2mm of travel while X shifts +1 mm. If I reverse the X_MIN stop logic, X goes in the right direction but fails to stop and crashes the connection to Octoprint. Attached are the current configs
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4562830/Configuration.zip)

ellensp, 2020-05-01:
check you endstop status with M119.  Manually triggering each endstop in turn with a finger test each endstop. Should be open when not triggered and triggered when it is triggered. 
Only invert endstop when it changes from open to triggered, but is in reverse. 
 
z sounds like its already triggered and just doing a re home.  so it homes (already home so does nothing) backs off (does this) home (oh already home do nothing) 

All of these are support questions. and this is not a support forum. It is for marlin firmware bugs and feature request only.

Please see https://github.com/MarlinFirmware/Marlin/issues/14694

closed 2020-05-02 by krawson14