MarlinFirmware/Marlin #17478: TMC 2130 connection error

ikoutone, 2020-04-11:
I have an ender 3 with mks gen 1.4 board and TMC2130 stepper drivers. I haven't been able to get it working with any of the Marlin 2.x versions (including latest 2.0.5.3). Currently I'm running Marlin 1.1.9, which works fine. So I don't know why I'm getting this connection error message with 2.x.

Here are my configuration.h, configuration_adv.h and pins_RAMPS.h files. Those are the ones I have edited. I copied pins and other changes I had made from my marlin 1.1.9 config
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4464243/Marlin.zip)

ellensp, 2020-04-11:
How about a M122 report and some images to make sure you wired it as your configuration says?

ikoutone, 2020-04-11:
I knew I forgot something, sorry. Wiring is a bit of a mess, but hopefully you can see something from the photos.

Send: M122
Recv: 		X	Y	Z	E
Recv: Enabled		false	false	false	false
Recv: Set current	800	800	800	800
Recv: RMS current	795	795	795	795
Recv: MAX current	1121	1121	1121	1121
Recv: Run current	25/31	25/31	25/31	25/31
Recv: Hold current	12/31	12/31	12/31	12/31
Recv: CS actual	31/31	31/31	12/31	12/31
Recv: PWM scale	255	255	0	2
Recv: vsense		1=.18	1=.18	1=.18	1=.18
Recv: stealthChop	true	true	true	true
Recv: msteps		0	0	16	16
Recv: tstep		4294967295	4294967295	max	max
Recv: PWM thresh.
Recv: [mm/s]
Recv: OT prewarn	true	true	false	false
Recv: off time	15	15	3	3
Recv: blank time	54	54	24	24
Recv: hysteresis
Recv:  -end		12	12	-1	-1
Recv:  -start		8	8	1	1
Recv: Stallguard thrs	0	0	0	0
Recv: DRVSTATUS	X	Y	Z	E
Recv: sg_result	1023	1023	0	0
Recv: stallguard	*	*
Recv: fsactive	*	*
Recv: stst
Recv: olb		*	*
Recv: ola		*	*
Recv: s2gb		*	*
Recv: s2ga		*	*
Recv: otpw		*	*
Recv: ot		*	*
Recv: Driver registers:
Recv: 		X	0xFF:FF:FF:FF	 Bad response!
Recv: 		Y	0xFF:FF:FF:FF	 Bad response!
Recv: 		Z	0x80:0C:00:00
Recv: 		E	0x80:0C:00:00
Recv: 
Recv: 
Recv: Testing X connection... Error: All HIGH
Recv: Testing Y connection... Error: All HIGH
Recv: Testing Z connection... OK
Recv: Testing E connection... OK

[image]
[image]
[image]

ellensp, 2020-04-11:
Ok, so only X and Y are not working...

ikoutone, 2020-04-11:
According to that, yeah. But actually testing the printer shows that E and X are not working. Can't extrude anything and printhead doesn't move sideways. Bed moves back and front.

ellensp, 2020-04-11:
No obvious pin issues with config files..

While off swap SPI cables from X and E stepper drivers. re run m122. has the fault moved from X to E or did it stay? If it moved you have a bad cable. 

While off now swap SPI cables from Y and Z stepper drivers and re run  m122. has the fault moved from Y to Z or did it stay? If it moved you have a second bad cable.

If the fault didn't move its is likely an issue with the stepper drivers themselves.

ikoutone, 2020-04-11:
Swapped cables from Z (which worked before) and E (which didn't work) and now the fault moved and I can't raise or lower Z but can extrude. So this means that my cable is bad. M122 shows exactly the same results as above.

ellensp, 2020-04-11:
yea,m122 wont change unless you also update CS pin to reflect new wiring...

Bob-the-Kuhn, 2020-04-11:
I've had problems crimping two wires into one contact.  Open connections and intermittent connections due to over crimping.  I eventually gave up and made a breakout board.

When that controller fried I bought a [BTT SKR_PRO](https://www.aliexpress.com/item/4000112172278.html?spm=a2g0o.productlist.0.0.2b81235br04Mb5&algo_pvid=d0bcae4a-5d91-4ac2-b945-c84ee0d54edc&algo_expid=d0bcae4a-5d91-4ac2-b945-c84ee0d54edc-3&btsid=0ab6f8ad15866242235285591e816a&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_) which has all the SPI wiring hardwired. So nice.  My only regret with that board is the SD card isn't mapped to the PC so I've been physically moving the SD card between my PC and the controller.

---

Right side - SDI, SCK & SDO from the controller.
Top - CS signals from the controller
Middle - 4 pin headers to the TMC chips

[image]

ellensp, 2020-04-14:
Please confirm you removed the 3 jumpers under each stepper driver. As these will interfere with SPI communications

ikoutone, 2020-04-14:
> Please confirm you removed the 3 jumpers under each stepper driver. As these will interfere with SPI communications

I have removed them. I have ordered a new cables from ali and waiting for them to arrive.

ikoutone, 2020-05-18:
To update this, I have received my new SPI cable from ali and I'm still getting the same TMC connection error. 

I didn't pay attention to this before, but for some reason there are no heatwaves above the heatbed in heatbed temp part of the screen. Not sure if this is a change in 2.x.x vs 1.1.9 where the heatbed has heatwaves above the bed.

But, looks like I'm forced to stay with marlin 1.1.9. Perhaps I'll try flashing 1.1.9.1, but 2.x series seems to be out of my reach.

boelle, 2020-06-20:
@ikoutone still an issue?

ikoutone, 2020-06-21:
> @ikoutone still an issue?

Yeah. I tried with Marlin 2.0.5.3 on May 18th, no luck then and I assume no luck now since no new version has been released.

boelle, 2020-06-21:
https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

updated just 10 hours ago

boelle, 2020-06-23:
Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

ikoutone, 2020-06-24:
> Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

Sadly it doesn't work. I'm still getting the same error message "TMC Connection error" on my printer's screen.

ikoutone, 2020-07-25:
Commenting because of stale label.

ikoutone, 2020-08-04:
I have just tested 2.0.6, I'm still getting the same TMC connection error message on my printer's screen.

Marsellus19, 2020-09-24:
Hi, did you have any luck? I have the same problem. Everything works fine under 1.1.9, but for some reason after updating to Marlin 2.0.6.1 the X and Y drivers cannot connect and the firmware reports an all high error. And I'm 100% positive it's not caused by wiring. 

`Recv: Driver registers:
Recv: 		X	0xFF:FF:FF:FF	 Bad response!
Recv: 		Y	0xFF:FF:FF:FF	 Bad response!
Recv: 		Z	0x80:0C:00:00
Recv: 		E	0x80:0D:00:00
Recv: 
Recv: 
Recv: Testing X connection... Error: All HIGH
Recv: Testing Y connection... Error: All HIGH
Recv: Testing Z connection... OK
Recv: Testing E connection... OK`

ikoutone, 2020-09-25:
Sadly not. I never got 2.x firmware working on my mks gen 1.4 board with tmc2130 so I'm stuck at 1.1.9.

ikoutone, 2020-09-26:
@ellensp @boelle any idea why it doesn't work? I got thinking "what is different between 1.1.9 and 2.x?" and I remembered that they use different tmcstepper libraries. I think my 1.1.9 uses TMC2130Stepper, while 2.x uses TMCStepper. Could this stepper library cause the issues I'm having?

ellensp, 2020-09-26:
What build environment are you using?
As Arduino ide will use what ever version of library you have installed
But Platformio will grab the version that platformio.ini tells it to, a version that is known to work. (TMCStepper@~0.7.1)

ellensp, 2020-09-26:
can you provide newer config files for the newer marlin.

ellensp, 2020-09-26:
Never mind, did it myself. 
[updated to bugfix Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5286335/updated.to.bugfix.Configuration.zip)

ikoutone, 2020-09-26:
I have used arduino ide in the past, but then I moved to visual studio code with platformio. 

I have no idea what I did differently this time, but I just tried marlin 2.0.6.1 and it works! All steppers are giving me OK when I run M122 and they also move correctly when I move them from octoprint.

ellensp, 2020-09-26:
Well that's good. as I thought I had some TMC 2130s laying about but don't.
Closing then.

closed 2020-09-26 by ellensp

Marsellus19, 2020-09-26:
I took me a lot of time and going back and forth between the two firmwares but I finally found the solution to the issue. under pins_RAMBO.h I had to change the X_CS_pin and Y_CS_pin to 59 and 63 respectively as that's where my drivers are hooked up. I don't remember ever changing that in Marlin 1.1.9 however the pins that I have my drivers plugged into are different to the default 2.0 ones. So I guess I must have changed them for some reason. Anyways. It works now. It turns out it kinda was a wiring issue after all, lol.

Koby88, 2020-11-03:
> I have used arduino ide in the past, but then I moved to visual studio code with platformio.
> 
> I have no idea what I did differently this time, but I just tried marlin 2.0.6.1 and it works! All steppers are giving me OK when I run M122 and they also move correctly when I move them from octoprint.

Hi i go exact same issue with MKS base 1.4 and tmc2130 i tried all but it works great on marlin 1.9 but not on 2.0 exact same error like yours... Can you pls somehow upload firmware that works for you so i can try it ?

Marsellus19, 2020-11-03:
> > I have used arduino ide in the past, but then I moved to visual studio code with platformio.
> > I have no idea what I did differently this time, but I just tried marlin 2.0.6.1 and it works! All steppers are giving me OK when I run M122 and they also move correctly when I move them from octoprint.
> 
> Hi i got exact same issue with MKS base 1.4 and tmc2130 i tried all but it works great on marlin 1.9 but not on 2.0 exact same error like yours... Can you pls somehow upload firmware that works for you so i can try it ?

Check your pins_RAMBO.h file in both firmwares. If your X and Y axis don't work then look what pins are assigned to X_CS_pin and Y_CS_pin in 2.0. If they're different to the pins specified in 1.9 just copy and paste. This should fix your problem :)

Koby88, 2020-11-03:
Why Rampo pins ? You think of RAMPS ? Rambo have nothing with mks sbase board

Marsellus19, 2020-11-04:
I guess that's what I meant 😆. If I recall correctly. In the pins file for the MKS, at the very top there was a #include pins_ramps.h or rambo.h. Although you're probably right. The mks is based on ramps.