MarlinFirmware/Marlin #17481: [BUG] Movements seem to be sometimes (randomly) skipping acceleration phase

hamster65, 2020-04-11:
Bug Description:
When I move x or y by sending commands like G1 X-10 F12000 most movements are smooth, but sometimes they produce a loud noise. It seems like the acceleration phase is missing. Sometimes this happens several times in sequence, sometimes it takes 40 of those G1 commands to reproduce the noise. It does not matter if there is a change of direction involved.

The same happens during G29 (bilinear leveling with bltouch). Sometimes, movements are smooth, sometimes they produce a loud and sudden noise. With lower currents, motors then stall.

I suspect this also happens during printing because sometimes I hear the same "thump!" noise.

Board is MKS SGEN L, drivers are TMC2300, corexy.
Configs attached.

Maybe someone can reproduce this?
Are there changes to my config that i could / should try?

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/4464416/configs.zip)

hamster65, 2020-04-11:
Odd. I just went to the TMC driver settings in the menu and set stealth chop to "off" of all motors.
Contrary to what I expected, the motors became more quiet... and the knocking noise is gone, so are the stalls and the shaking.

Hagbard-84, 2020-04-11:
Related? #17473

hamster65, 2020-04-12:
No, I don't think this is related.
I realized that #define HYBRID_THRESHOLD was not enabled. So I was running on stealth chop the whole time! Considering that I printed 150mm/s with this setting.... :)... 
So the "missing acceleration phase" was probably simply caused by a lack of torque and missed steps.

However, now with HYBRID_THRESHOLD enabled, it immediately stalls when I set the threshold somewhere above 25. I also tested with the thresholds set to max (255). 

Looks like it has less torque when  HYBRID_THRESHOLD is enabled and thresholds are set to 255 and more torque with HYBRID_THRESHOLD disabled and stealth mode active. 
But from my understanding, this should not make any difference.

hamster65, 2020-04-12:
ok... there is definitely something going on here. I tested with different threshold settings for the the Z axis.

0:  ok
1: quieter, slight knocking noise on start and stop
2-4: same noise, but louder knocking noise
5: stalls / missing steps in a repeating pattern (moves, but klack klack klack noise)
6-255: ok, sounds exactly like setting 0

boelle, 2020-06-20:
@hamster65 is issue gone?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle