MarlinFirmware/Marlin #17490: Pull request

AntoszHUN, 2020-04-11:
Hello.
I can't send a pull request. I would like to put the Hungarian language in Marlin. But this error appears during the check. Can you give an idea of advice as to why it is not good?

[image]

Thanks in advance for the help! I apologize for the inconvenience!

thisiskeithb, 2020-04-11:
Your branch needs to start with `bugfix-2.0.x` and choose `bugfix-2.0.x` as the base when doing the PR instead of `2.0.x`.

We don’t merge directly into `2.0.x`.

AntoszHUN, 2020-04-11:
Thanks! I try.

AntoszHUN, 2020-04-11:
Thanks for the help! It looks like bugfix-2.0.x will be good. still checking. I'm sleeping. Can I know when it will be included in the Firmware? Which version? Because now this 2.0.x or bugfix-2.0.x is a bit confusing. See also 2.0.5.3. Because what we downloaded is 2.0.5.3, right? Thanks for the reply in advance. I'll close after that.

thisiskeithb, 2020-04-11:
> Can I know when it will be included in the Firmware?

If everything looks good, likely soon.

> Which version? Because now this 2.0.x or bugfix-2.0.x is a bit confusing. See also 2.0.5.3. Because what we downloaded is 2.0.5.3, right? 

Updates and code fixes go into `bugfix-2.0.x` which will eventually get merged into `2.0.x` as a new release. See [Contributing Code with Pull Requests](https://marlinfw.org/docs/development/getting_started_pull_requests.html) for more info.

AntoszHUN, 2020-04-11:
Oooo I get it already! Thanks! All the best!

closed 2020-04-11 by AntoszHUN