MarlinFirmware/Marlin #17493: POWER_LOSS_REY enabled but Outage Recovery defaults to Off

nicwilson58, 2020-04-12:
If POWER_LOSS_RECOVERY is enabled.   Configuration->Outage Recovery defaults to Off,  is this normal,  what needs to be done to  make Outage Recovery default to On

thisiskeithb, 2020-04-12:
Change `PLR_ENABLED_DEFAULT` to `true`.

thinkyhead, 2020-04-12:
Turn it ON, then send `M500`.

thinkyhead, 2020-04-12:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

closed 2020-04-12 by thinkyhead