MarlinFirmware/Marlin #17494: [BUG] Relationship between NOZZLE_TO_PROBE_OFFSET and UBL's MESH_MAX_X/Y not as clear as it could be

evilC, 2020-04-12:
### Bug Description
Apologies for using Bug for this, I didn't really see a more appropriate category though - it's not really a "Feature Request" IMHO

When trying to set up UBL, `G29 P1` would fail with "Probing failed" (Despite the fact that it did not even attempt to probe the point that it failed on - ie it did not deploy my BLTouch's probe for that point or even move to the point - it moved to the home position) towards the end of the probing, and even turning verbosity up to max yielded no real clues as to the problem.  
In the end, I finally noticed that it was always failing on the last row or last column, which finally led me to the solution
The issue was that it was trying to move outside the limits defined for the bed.
It was only once I set `MESH_MAX_X` and `MESH_MAX_Y` to compensate for `NOZZLE_TO_PROBE_OFFSET` that it finally started working

Ideally, what I would have liked to have done was this:
```
  #define MESH_MAX_X X_BED_SIZE - ((NOZZLE_TO_PROBE_OFFSET[0] * -1) + MESH_INSET)
  #define MESH_MAX_Y Y_BED_SIZE - ((NOZZLE_TO_PROBE_OFFSET[1] * -1) + MESH_INSET)
```
But it did not seem to like that syntax

So in the end I did
Configuration.h
```
// Add custom defines NOZZLE_TO_PROBE_OFFSET_X and NOZZLE_TO_PROBE_OFFSET_Y so they can be used in MESH_MAX_X / MESH_MAX_Y
#define NOZZLE_TO_PROBE_OFFSET_X -31
#define NOZZLE_TO_PROBE_OFFSET_Y -35
#define NOZZLE_TO_PROBE_OFFSET { NOZZLE_TO_PROBE_OFFSET_X, NOZZLE_TO_PROBE_OFFSET_Y, -2.65 }
```

Configuration_adv.h
```
  // Use custom NOZZLE_TO_PROBE_OFFSET_X and NOZZLE_TO_PROBE_OFFSET_Y defines to calculate MESH_MAX_X and MESH_MAX_Y
  #define MESH_MAX_X X_BED_SIZE - ((NOZZLE_TO_PROBE_OFFSET_X * -1) + MESH_INSET)
  #define MESH_MAX_Y Y_BED_SIZE - ((NOZZLE_TO_PROBE_OFFSET_Y * -1) + MESH_INSET)
```

TLDR at the very least I feel that there should be a comment in `Configuration_adv.h` to indicate that  `MESH_MAX_X` and `MESH_MAX_Y` need to take into account `NOZZLE_TO_PROBE_OFFSET` - if I, as a coder has trouble figuring it out, less technical people will probably have a much harder time.
About the closest thing I could find to a hint about this is the comment on [this page](https://marlinfw.org/docs/features/unified_bed_leveling.html) that says `Note though that (unlike bilinear leveling) the MESH_MIN and MESH_MAX positions refer to where the nozzle can reach, not the probe`

Thoughts?

qwewer0, 2020-04-12:
(If I'm not wrong) You can get all that by changing the X_MAX_POS/Y_MAX_POS to (X_BED_SIZE/Y_BED_SIZE + K/L), where K/L is the printers ability to move beyond the bed size in X/Y direction. But not all printer can go further than the bed size.

evilC, 2020-04-12:
The reason I raised the issue is not that I can't do something that I want to, it's that it was unclear WHY the probe was failing, and searching the internet did not yield any useful info

I didn't want to probe more area of the bed than could be done automatically, I just wanted to get the automatic probing to probe what it could, and it was not clear to me initially why it was failing to complete. 

Subtracting the probe offsets from MESH_MAX_X and MESH_MAX_Y allowed the automatic probing to complete, and yielded a large enough mesh for my needs.

The issue was raised to highlight a (perceived) shortcoming of the documentation and attempt to stove off support requests from less technically-minded people who don't figure out what the true cause of the issue is. Hell, maybe another idea would be to throw a more useful error such as "Cannot move outside printable area" rather than just an unhelpful "Probing failed". Technically, probing did NOT fail, it was a move that failed!

To address your suggestion, surely X_MAX_POS and Y_MAX_POS control where the nozzle can move during a print? Setting these to values which would allow the nozzle to move off the bed (And potentially the carriage crashing into things) in order to probe those points does not seem like a sensible workaround

qwewer0, 2020-04-12:
Got it.
"Cannot move outside printable area" would be a good idea.

I moved my X carriage by hand to see how far can it go beyond the X_MAX_POS, and got X_BED_SIZE + 30, so I changed it accordingly.  After that I was able to probe all the mesh points.
But as I mentioned, not all printer can go further than the bed size.

thinkyhead, 2020-04-12:
The probe offsets can be changed during runtime too.

By way of documentation, we simply say that UBL "probes all the points that it can reach according to the settings you have provided." If it *looks like* UBL should be able to probe all points but it can't then it is most likely the MIN_POS and MAX_POS values are too narrow.

So you did the right thing.

closed 2020-04-12 by thinkyhead