MarlinFirmware/Marlin #17495: Build error for SKR Pro V1.1

xeddog, 2020-04-12:
### Bug Description

Build error as shown in the attached screenshot.
[image]

### My Configurations

[Configuration files.zip](https://github.com/MarlinFirmware/Marlin/files/4465751/Configuration.files.zip)

### Steps to Reproduce
Just run the build.  

**Expected behavior:** [What you expect to happen]

Successful Build

#### Additional Information
This is for an SKR Pro V1.1 With TMC2209 drivers for X, Y, Z, and Z2, and an LV8729 for E.  The source compiled Marlin 2.0.5.2 without issue, and that is what is currently running in the printer.  I ran a pull two days ago (should be to the 2.0.5.5 level), and the error has appeared and is repeatable.  There have been no changes to the Marlin source except the pull.  No changes by me.  I have made sure that VSCode and Platformio are all up-to-date, as well as all extensions and packages.  I have also run Clean prior to trying the build, and reset Intellisense data base as suggested on the facebook forum.  Lastly, I deleted the .pio and .vscode directories and tried again with the same result.

sjasonsmith, 2020-04-13:
Your configs build for me out of the bugfix-2.0.x branch. The picture you show reflects an old version of the Arduino framework. Updating that may resolve your problems.

I believe we had the versions specified incorrectly in the platformio.ini. I've posted a pull request to update that, so that hopefully the proper minimum version (3.10700) would be required for your build.

xeddog, 2020-04-13:
> 
> 
> Your configs build for me out of the bugfix-2.0.x branch. The picture you show reflects an old version of the Arduino framework. Updating that may resolve your problems.

I think I found it.  Not sure why, but . . . 

I opened platformio (clicked on the little space dude), and looked at the platforms that were installed.  There was three ST STM32 versions installed.  Versions 5.6.0, 5.4.3, and 6.1.0.  I un-installed 5.4.3 and 5.6.0, ran a clean, and when the clean started it downloaded and installed the 3.107 framework needed.  I then ran a build and it completed successfully.

closed 2020-04-13 by xeddog