MarlinFirmware/Marlin #17504: [BUG] Testing X Y Z E connection... Error: All HIGH after heating

superkefir, 2020-04-12:
### Bug Description
Hello!
Last week i have a battle with my TMC5160.
On boards: SKR 1.3(2pcs) and SKR 1.4 Turbo
I got the same problem with TMC5160 1.2 from BTT (X Y Z E Z2)
Version of Marlin 2.0.5.2->2.0.5.3

When power on the board and send M122 i have the message that connection is ok and the register of the drivers.
After sending M190 S90 (or M109 S220 T0) the printer sending busy in terminal and after that if i send M122 i have connection error: all high for all drivers until i reboot the board.
It can prints but can't change threshold values and prints very loud. Before sending one if this commands it moves in all positions without any problems and M122 respond good for all drivers.
If i heat the bed and the extruder by sending M140 S90 and M104 S220 T0. The response is good.
### My Configurations
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4466554/Marlin.zip)

### Steps to Reproduce

1) Turn on
2) Send M122
3) Send M190 Sx or M109 Sx T0
4) Wait until it heats
5) Send M122

#### Additional Information

[image]
[image]
[image]

GMagician, 2020-04-12:
Maybe some electrical noise on spi wires due to heater pwm?

superkefir, 2020-04-12:
But the heater works fine when i send M140 Sx (from 30 to 100). It's just heat and then M122 is normal. 
Also i disconnect all TMC5160 and check with only one on each axis. It's the same situtation. 
Also i checked with only extruder heater and only bed heater that is 220v and connected through the solid-state relays.

superkefir, 2020-04-12:
I'm confused. After M190 Sx i send M122, then i see bad response and send M502 and M500 and again M122. And the connection is back.
If i power on the board and send M502 -> M500 -> M190 Sx the problem is the same. If after M190 Sx a send M502 -> M500 and then try M190 Sx or M109 Sx T0 the connection is ok.

thinkyhead, 2020-04-12:
When the board is "powered on" does it have the full 12V? The 12V power is needed for TMC drivers to be able to accept and retain settings.

superkefir, 2020-04-12:
I checked with multimeter it has 24v. The heater of extruder when it turns on the same ~24v.

superkefir, 2020-04-14:
I have some idea.
I can't check LPC1769 on version below 2.0.5.0
So i check the version 2.0.5.0, 2.0.5.1 2.0.5.2
Everything the same.
But have some consistent patterns.
So my steps:
1) Turning the board
2) Check connections of driver by M122
3) It's ok so i load factory default by M502 and M500 to save it.
4) M190 S30 (or other temperature, it doesn't matter) 
5) Waiting for response and send M122 - getting connection lost
6) Send M502 or M501 - it's resume connection.
7) Send M190 Sx
8) Waiting and send M122 again - everything ok.

So the idea that the difference between commands M190 and M140 do the same temperature, but the second command doesn't broke the connection and the first does. But connection turns back after restoring factory defaults. I open the file printcounter.h to check what can broke the EEPROM settings. But by knowledge of the Marlin is not enough to find the problem. Only possible reasons:

> #if USE_REAL_EEPROM
>   // round up address to next page boundary (assuming 32 byte pages)
>   #define STATS_EEPROM_ADDRESS 0x40
> #else
>   #define STATS_EEPROM_ADDRESS 0x32
> #endif

or maybe

>  #if USE_REAL_EEPROM || defined(CPU_32_BIT)
>       typedef uint32_t eeprom_address_t;
>     #else
>       typedef uint16_t eeprom_address_t;
>     #17059 
But i can't find a working version for LPC1769 that won't broke the connection (or EEPROM data) so maybe it will helps to find what's the reason of such behavior.
But maybe you can help me to find how to fix it without buying fourth board of SKR.

GMagician, 2020-04-14:
You may send m503 to see current values. Comparing them with before and after heating you may see what's different. But I think it's not something related to eeprom corruption but something else. I can't dig more because I don't have a SKR board.
Something you may do is an "M43" (not sure if with W argument) before and after heating, just to see if some pin has changed its "I/O" state.

N.B. M43 requires firmware to be compiled with `PINS_DEBUGGING`

superkefir, 2020-04-14:
I only see that the pins are changed but i knew it before, cause there is no connection. But what can broke in settings by command M190 that doesn't by M140.
[M122before.txt](https://github.com/MarlinFirmware/Marlin/files/4477912/M122before.txt)
[M122after.txt](https://github.com/MarlinFirmware/Marlin/files/4477913/M122after.txt)
[M503 before.txt](https://github.com/MarlinFirmware/Marlin/files/4477914/M503.before.txt)
[M503 after.txt](https://github.com/MarlinFirmware/Marlin/files/4477915/M503.after.txt)
[M43 before.txt](https://github.com/MarlinFirmware/Marlin/files/4477916/M43.before.txt)
[M43after.txt](https://github.com/MarlinFirmware/Marlin/files/4477917/M43after.txt)

GMagician, 2020-04-15:
TMC_SW_SCK is 1 when ok and 0 when not, that's not good. Even in case of error clock should be in idle state on end of communication

superkefir, 2020-04-15:
Is it possible that the M190 command when stops the command to printer may depend on the other work of printer that's broke connection or can't return some functions back? At the same time M140 works good. And after M501 when the connection is broken (or the connection is working but some other settings are broken, that shows that connection is lost) it starts working normal and the M190 command doesn't brake it again.

superkefir, 2020-04-15:
It seems like TMC_SW_SCK is 0 after M501 command.

GMagician, 2020-04-15:
m501 calls `reset_stepper_drivers` and there I think it reprogram TMC_SW_SCK.

First of all it's important to know if in your board is used software or hardware SPI...

superkefir, 2020-04-15:
I mean after the M190 the command M43 shows TMC_SW_SCK 0 and M501 doesn't change it back to 1.

GMagician, 2020-04-15:
Both M140 and M190 call `restore_stepper_drivers` when AUTO_POWER_CONTROL is defined and M190 just wait temp is reached. No other differences with M140.
You said  that M140 has no problems. What if you call M140, wait bed temperatures reached and then M122?

superkefir, 2020-04-15:
It's all ok.
You can check the attached treminal info with M43 -> M140 S30 -> M43 and M122
[M140.txt](https://github.com/MarlinFirmware/Marlin/files/4483296/M140.txt)

superkefir, 2020-04-15:
The same situation with M104 and M109. M104 it's ok and M109 broke the drivers.
[M104_M109.txt](https://github.com/MarlinFirmware/Marlin/files/4483320/M104_M109.txt)

GMagician, 2020-04-15:
Do you have some led lighting (PRINTER_EVENT_LEDS or similar) ?
TMC_SW_SCK and RGB_LED_G_PIN seems to share same pin

superkefir, 2020-04-15:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4483359/Marlin.zip)
I have 

```
#define RGB_LED
#if EITHER(RGB_LED, RGBW_LED)
  #define RGB_LED_R_PIN 5
  #define RGB_LED_G_PIN 4
  #define RGB_LED_B_PIN 6
  #define RGB_LED_W_PIN -1
#endif
```

superkefir, 2020-04-15:
oh

GMagician, 2020-04-15:
Try to disable it...

closed 2020-04-15 by superkefir