MarlinFirmware/Marlin #17506: [QUESTION] VSCode warnings at bugfix-2.0.x ac7b484

qwewer0, 2020-04-12:
I can successfully compile the firmware, but I get these warnings, and don't know if I should do something about them:
```
{
	"resource": "/c:/Users/qwewer/Documents/Marlin/Marlin/src/lcd/menu/menu_ubl.cpp",
	"owner": "cpp",
	"severity": 4,
	"message": "'%i' directive writing between 1 and 6 bytes into a region of size between 5 and 10 [-Wformat-overflow=]",
	"startLineNumber": 189,
	"startColumn": 6,
	"endLineNumber": 189,
	"endColumn": 6
}
```
```
{
	"resource": "/c:/Users/qwewer/Documents/Marlin/Marlin/src/module/stepper/trinamic.cpp",
	"owner": "cpp",
	"severity": 4,
	"message": "unused variable 'sp_helper' [-Wunused-variable]",
	"startLineNumber": 343,
	"startColumn": 7,
	"endLineNumber": 343,
	"endColumn": 7
}
```
<details>
  <summary>Terminal output</summary>

  ```
  > Executing task: C:\Users\qwewer\.platformio\penv\Scripts\platformio.exe run <

Processing STM32F103RC_btt_512K (platform: ststm32; board: genericSTM32F103RC; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 6.1.0 > STM32F103RC (48k RAM. 256k Flash)    
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 512KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32-maple 2.10000.200103 (1.0.0)        
 - tool-stm32duino 1.0.2
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 39 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <LiquidCrystal>
|-- <TMCStepper> 0.6.2
|   |-- <SoftwareSerialM> 1.0.0
|-- <U8glib-HAL> 0.4.1
|   |-- <Wire> 1.0
|-- <Adafruit MAX31865 library> 1.0.3
|-- <LiquidTWI2>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.8.0
|-- <SailfishLCD>
|-- <SlowSoftI2CMaster>
|-- <SoftwareSerialM> 1.0.0
|-- <STM32ADC> 1.0
|-- <EEPROM>
|-- <USBComposite for STM32F1> 0.91
|-- <Wire> 1.0
Building in release mode
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduinoVariant\board.cpp.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduinoVariant\wirish\boards.cpp.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduinoVariant\wirish\boards_setup.cpp.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduinoVariant\wirish\start.S.o
[... 758 lines cut ...]
jgaurora_a5s_a1           IGNORED
STM32F103CB_malyan        IGNORED
chitu_f103                IGNORED
STM32F401VE_STEVAL        IGNORED
FLYF407ZG                 IGNORED
FYSETC_S6                 IGNORED
STM32F407VE_black         IGNORED
BIGTREE_SKR_PRO           IGNORED
BIGTREE_GTR_V1_0          IGNORED
BIGTREE_BTT002            IGNORED
teensy31                  IGNORED
teensy35                  IGNORED
esp32                     IGNORED
linux_native              IGNORED
SAMD51_grandcentral_m4    IGNORED
rumba32_f446ve            IGNORED
rumba32_mks               IGNORED
include_tree              IGNORED
==================================================================================== 1 succeeded in 00:02:25.550 ====================================================================================
  ```
</details>

Ender 3
SKR Mini E3 v1.2
[Configuration files.zip](https://github.com/MarlinFirmware/Marlin/files/4467052/Configuration.files.zip)

thisiskeithb, 2020-04-12:
Warnings can be ignored. Only worry about errors, which will halt compilation.

closed 2020-04-12 by qwewer0

sjasonsmith, 2020-04-13:
> Warnings can be ignored. Only worry about errors, which will halt compilation.

The warning about the destination buffer size is concerning. I'll put a fix up for it in a little bit.