MarlinFirmware/Marlin #17509: [BUG] Config example

chepo92, 2020-04-13:
### Bug Description

not excatly a bug but in release 2.0.5.3, all the examples config are gone and there this file with this link that goes nowhere: 
Where have all the configurations gone?
https://github.com/MarlinFirmware/Configurations/archive/release-2.0.5.2.zip

### My Configurations

There is no config! 

### Steps to Reproduce

1. Download release
2. look for example config
3. open file and link

**Expected behavior:** the configs

**Actual behavior:**

(Returns 404)

#### Additional Information
I understand i need to go to release 2.0.5.2
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

chepo92, 2020-04-13:
and they are not in release 2.0.5.2, where then?

chepo92, 2020-04-13:
Why it has been broken many times? 

I also will improve my search words selection as i didn't find the other issues :D

thinkyhead, 2020-04-27:
https://github.com/MarlinFirmware/Configurations/archive/release-2.0.5.zip

closed 2020-04-27 by thinkyhead