MarlinFirmware/Marlin #17511: [BUG ]  2.0.5.3 Compiling for Mightyboard fails

chepo92, 2020-04-13:
### Bug Description
I'm compiling for a Mightyboard with default configs provided for FlashForge Creator, 

C:\Users\Axel\AppData\Local\Temp\arduino_build_382875\sketch\src\feature\digipot\digipot_mcp4018.cpp:28:10: fatal error: utility/twi.h: No such file or directory

ResolveLibrary(utility/twi.h)
  -> candidates: [] #include "utility/twi.h"

### My Configurations
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/4467995/configs.zip)
**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. downloaded lastest release 2.0.5.3
2. copy configs to folder
3. try to compile and get the error
Notice:  you need to have library SlowSoftI2C:
#include <SlowSoftI2CMaster.h>  //https://github.com/stawel/SlowSoftI2CMaster

**Expected behavior:** 
compiles!

**Actual behavior:** 
fatal error: utility/twi.h: No such file or directory

#### Additional Information

As i stated in #12855, comenting line 28 in src\feature\digipot\digipot_mcp4018.cpp
//#include "utility/twi.h"
will fix the issue, i would PR the fix to bugfix branch but i don't know if that would break compiling with other configs or if the error is just happening to me

sjasonsmith, 2020-04-13:
This builds fine for me in the bugfix-2.0.x branch. I did not try the released branch.

`utility/twi.h` comes from the Arduino AVR framework. If it cannot be found there may be something wrong with the copy PlatformIO has cached on your system.

darronr, 2020-04-13:
I had the same issue using the Arduino IDE.  
I commented out the line to see what would fail and nothing did.
NOTE in "bugfix" this has been changed to <utility/twi.h>  - this also failed to compile.

sjasonsmith, 2020-04-13:
Oh, you are all just running into path length issues on Windows if it isn’t working in Arduino. Move your source to a shorter path and try again. Try something like “C:\Marlin”.

darronr, 2020-04-13:
@sjasonsmith that is not the issue.
twi.h is not used anywhere else and it compiles with the line removed - try it in VS
I believe that it's trying to access methods that are included in "SlowSoftI2CCMaster.h"
**So this line should just be removed.**

class SlowSoftI2CMaster {
 public:
  SlowSoftI2CMaster(uint8_t sda, uint8_t scl);
  bool i2c_init(void);
  bool i2c_start(uint8_t addr);
  bool i2c_rep_start(uint8_t addr);
  void i2c_stop(void);
  bool i2c_write(uint8_t value);
  uint8_t i2c_read(bool last);
  bool error;
  
 private:
  uint8_t _sda;
  uint8_t _scl;
};

chepo92, 2020-04-13:
I'm using Arduino IDE
Neither combination of using <utility/twi.h> and/or moving to C:\Marlin worked
utility/twi.h is located in Arduino\hardware\arduino\avr\libraries\Wire\src, so it should be automatically included when Wire library is included
so I tried moving utility/twi.h to 
Arduino\hardware\arduino\avr\libraries\  didn't work
Marlin\src\feature\digipot   worked! 
further investigating the digipot library example (included in Wire library) doesn't have the #include "utility/twi.h" but just including the Wire.h 

Maybe this line is unnecesary ?

chepo92, 2020-04-13:
> 
> 
> @sjasonsmith that is not the issue.
> twi.h is not used anywhere else and it compiles with the line removed - try it in VS

Yes, is not used (almost with this config), that is why commenting it out worked

> I believe that it's trying to access methods that are included in "SlowSoftI2CCMaster.h"

Yes it is

> **So this line should just be removed.**

Agree, but are we sure this won't break anything else (with other boards using digipots?)

That's why i'm asking here:
> i would PR the fix to bugfix branch but i don't know if that would break compiling with other configs or if the error is just happening to me

darronr, 2020-04-13:
@chepo92 if that is the case then the following line 
`#include <SlowSoftI2CMaster.h>`
would break non-mightyboard builds that use I2C and MCP4018 anyway.  This class is meant to use the sailfish library version and not the avr one.

Looking at config_adv.h - mightyboard is the only one to use MCP4018
```
   *                        A   (A shifted)   B   (B shifted)  IC
   * Smoothie              0x2C (0x58)       0x2D (0x5A)       MCP4451
   * AZTEEG_X3_PRO         0x2C (0x58)       0x2E (0x5C)       MCP4451
   * AZTEEG_X5_MINI        0x2C (0x58)       0x2E (0x5C)       MCP4451
   * AZTEEG_X5_MINI_WIFI         0x58              0x5C        MCP4451
   * MIGHTYBOARD_REVE      0x2F (0x5E)                         MCP4018
```

So the line removal should be safe.

closed 2020-04-14 by thinkyhead