MarlinFirmware/Marlin #17514: [BUG] LCD 2004 Mightyboard

chepo92, 2020-04-13:
### Bug Description
I'm using updating some mightyboards with stock LCD to marlin and facing this issues:
In some points navigating the menu (eg. moving axes), there appears a weird "H" character in the LCD. 

Also in some parts it shifts one char position. 

### My Configurations
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/4468391/configs.zip)

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Upload
2. Navigate through menu

**Expected behavior:** clear lcd without rare chars

**Actual behavior:** "H"

#### Additional Information
[image]

[image]

[image]

[image]

ellensp, 2020-04-13:
Did you use official marlin? The reason I ask is the bug that places the last line one char right, I remember fixing a long time ago...  unless its regressed..

chepo92, 2020-04-13:
I'll give a try to bugfix branch

boelle, 2020-06-20:
@chepo92 is the issue still there

chepo92, 2020-06-20:
> 
> 
> @chepo92 is the issue still there ?

Yes, I used bugfix a month ago and it was still present, I haven't done much debugging/reserach on it as is not critical

boelle, 2020-06-23:
Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

its updated daily and it might well be that your issue have been solved

boelle, 2020-06-29:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

chepo92, 2020-07-11:
Tested bugfix today, issue still present, any advice? I'm guessing is something related to LCD, timings or libraries

[image]

[image]

[image]

[image]

chepo92, 2020-07-11:
I'm also having #17515 with this LCD and board

closed 2020-07-14 by thinkyhead