MarlinFirmware/Marlin #17515: [BUG] Malin 2.0 Digits missing in move axis Z with step of 0.025 LCD2004

chepo92, 2020-04-13:
### Bug Description
setup: Ramps and LCD 2004, marlin 2.0.5.3, 
using move axis in the LCD with a step of 0.025 to position Z in a 2 or 3 integer digit  (say 10, 100) it will be missing the "+" or "-" character and the most significant digits until the units (if it is in +12.345 or +112.345 it will show 2.345) , 

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4468510/Marlin.zip)

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Use move axis (use with large steps, say 10) to put Z in a 2 or 3 digit position (eg 12.00, 123.00)
2. Go back to step selection and select 0.025 step, it will show only 2.000 or 3.000 

**Expected behavior:** it has plenty of chars space so display all digits

**Actual behavior:** trims the number to the unit and decimals

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

boelle, 2020-06-20:
@chepo92 is the issue still there?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

chepo92, 2020-07-11:
@boelle yes the issue is still there, tested on lastest bugfix-2.0, please keep it open, as I'll try to solve it as no one else is working here (any advice is welcome)

thinkyhead, 2020-07-13:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

chepo92, 2020-07-13:
Confirmed, just downloaded bugfix 2.0 from marlin repo, it is somehow limiting to show 4 digits and a decimal point (x.xxx) in move 0.025 steps mode, not showing the sign, cents o tens. 

[image]

[image]

closed 2020-07-14 by thinkyhead

chepo92, 2020-07-14:
No need to test? :D

thinkyhead, 2020-07-14:
I tested on two printers. Seems fine, and shouldn't vary with other systems.