MarlinFirmware/Marlin #17516: [BUG] REPRAP_DISCOUNT_SMART_CONTROLLER doesn't work on BOARD_MELZI_CREALITY board

DominikPalo, 2020-04-13:
### Bug Description
I replaced my Ultimachine Mini-Rambo board with a Creality silent board, but it seems that board works only with a stock Creality LCD, not with RepRap Discount smart controller LCD.

### My Configurations
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4468586/configuration.zip)

### Steps to Reproduce
1. Use an [example Creality Ender 3 configuration](https://github.com/MarlinFirmware/Configurations/tree/release-2.0.5/config/examples/Creality/Ender-3)
2. Comment out `#define CR10_STOCKDISPLAY` and uncomment `#define REPRAP_DISCOUNT_SMART_CONTROLLER`
3. Connect a RepRap Discount smart controller to the board (just the AUX1 cable)
4. Turn on the board

I'm using Marlin 2.0.5

**Expected behavior:** [What you expect to happen]
The connected RepRap Discount smart controller works.

**Actual behavior:** [What actually happens]
The buzzer and the reset button work, but the display doesn't show anything (just a thin horizontal line)

ellensp, 2020-04-13:
This is not a bug, it is a limitation of the hardware. REPRAP_DISCOUNT_SMART_CONTROLLER is just not compatible with this controller.  Yes the encoder and the sdcard are on the second connector. So in theory you could use only the display But the pin out of exp1 is not compatible.

DominikPalo, 2020-04-13:
Thank you, I wasn't aware of that - so in this case, I think there should be a warning during the compilation (something like the BIGTREE_SKR_MINI_E3 board has: 

`#error "Only ZONESTAR_LCD, MKS_MINI_12864, ENDER2_STOCKDISPLAY, and CR10_STOCKDISPLAY are currently supported on the BIGTREE_SKR_MINI_E3."`

sl1pkn07, 2020-04-13:
or you can make a adapter

DominikPalo, 2020-04-13:
@sl1pkn07 I know that the Creality controller somehow merges the LCD, the speaker and the encoder into a single 10-pin EXP3 connector, but when I look at the RepRap Discount controller schematics (attached), it needs 3 pins for encoder (2 of them are on the EXP2). But the LCD and the speaker use all pins on the EXP1. So I'm not sure how can I connect all devices using just 10 pins, because 1x Vcc pin, 1x Gnd pin, 3x encoder pin, 1x speaker pin and 6x LCD pin = 12 pins (I intentionally omitted another 1x Reset pin from EXT2 - Creality board doesn't have the Reset button and I don't need it). Maybe some of the LCD pins are not required, but unfortunately, I was not able to find any schematic for Creality controller not schematic of the adapter. Could you please help?

[image]

sl1pkn07, 2020-04-13:
this is my adapter from CR10 display to Rearm

~~~
EXP3-01 ~ BEEP       -> J3-03 ~ P2.11 =  Amarillo (cable gordo)
EXP3-02 ~ ENC_BUTTON -> J3-04 ~ P3.25 =  Amarillo
EXP3-03 ~ ENC_B      -> J3-05 ~ P1.23 =  Marron
EXP3-04 ~ ARD_RST
EXP3-05 ~ ENC_A      -> J3-07 ~ P0.16 =  Blanco
EXP3-06 ~ LCD_SCK    -> J3-08 ~ P2.06 =  Naranja
EXP3-07 ~ LCD_CS     -> J3-09 ~ P0.15 =  Azul
EXP3-08 ~ LCD_MOSI   -> J3-10 ~ P0.18 =  Verde
EXP3-09 ~ GND        -> J3-11 ~ GND   =  Negro
EXP3-10 ~ VCC        -> J3-12 ~ +5V   =  Rojo
~~~

the lcd only use 3 (SCK, CS and MOSI)

this video can be use like example

ellensp, 2020-06-15:
Not a firmware related, please close this issue

closed 2020-06-21 by boelle