MarlinFirmware/Marlin #17517: [BUG] (short description)

Gerggg, 2020-04-13:
### Bug Description

I updated my printer yesterday, and today I report that the BLtouch no longer works during the G28. The head does not stop and pierced my bed ...

I saw that there were several issues on the same problem but I cannot find the origin of the problem.

### My Configurations

My config:
SKR 1.4 turbo
Bltouch
TMC2209 Uart

### Steps to Reproduce

1. make G28
2. cry for your bed

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4468724/Marlin.zip)

superkefir, 2020-04-13:
Did you check [this ](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/261#issuecomment-606955654) solution from BTT tree?

thisiskeithb, 2020-04-13:
There’s an open PR to help with boards like the SKR 1.4 that have a dedicated probe pin since Marlin currently assumes you’ll use Z-min instead: https://github.com/MarlinFirmware/Marlin/pull/17359

Gerggg, 2020-04-13:
the problem comes from a wire cut at the level of the skr

closed 2020-04-13 by Gerggg