MarlinFirmware/Marlin #17519: MKS RUMBA32 FAN M106 open it, incorrect voltage output

vivary, 2020-04-13:
### Bug Description
1. Use the old version of marlin2.0, the voltage output to RUMBA32, FAN is normal;
2. Use Marlin2.05 firmware, compile and download it to the RUMBA32 motherboard through platformio, use the command M106 P0 S255 to control to open FAN, but the output voltage of the FAN interface is only 2.2V, the normal voltage should be to the power supply voltage
[image]
[image]

### My Configurations
[Marlin-RUMBA32.zip](https://github.com/MarlinFirmware/Marlin/files/4469016/Marlin-RUMBA32.zip)
This is my config.

### Steps to Reproduce
1. Upload marlin2.0.5 to RUMBA32
2. Use pronterface send command to RUMBA32 board by USB
3. Measure FAN output voltage

#### Additional Information
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-04-13:
The mosfets are on the low side of the cct.  Ie the +v is directly connected to the power supply.
The gnd is switched. 
To measure the voltage you have to use the GND pin of the fan plug as your meter gnd. 
See https://github.com/Aus3D/RUMBA32/blob/master/Schematic.pdf

vivary, 2020-04-13:
RUMBA32 motherboard, sending M106 P0 S255, the voltage output is abnormal, which may be caused by the HAL library timer or other causes.

ellensp, 2020-04-13:
Is your 12v there?
What is the voltage at the mosfet gate compared with real gnd.
Can you put a scope on the gate see if its is PWMing or constant high.

makerbase-mks, 2020-04-16:
[image]
This setting can solve the problem.

sjasonsmith, 2020-05-29:
It works if you comment the following three lines out of the variant file:
```
// PWM resolution
//#define PWM_RESOLUTION          8
//#define PWM_FREQUENCY           20000 // >= 20 Khz => inaudible noise for fans
//#define PWM_MAX_DUTY_CYCLE      255
```

This causes it to run fine, but with only a 1 kHz PWM frequency. I believe this is how the SKR Pro runs right now also, and has been the subject of some complaints.

There have been some changes involving PWM in more recent frameworks. The 20 kHz PWM seems to work fine in stm32duino 1.8, but some other issues are preventing that upgrade right now.

Ironically, with the fan I am using 1 kHz PWM works great, but the intended 20 kHz PWM is unusable, even though it looks perfect on the scope. I would have to use FAN_SOFT_PWM or perhaps add capacitors for the higher frequency PWM to be useful at all.

On another note, there is a timer conflict on this board between the TONE timer and the STEP timer. That will cause problems if anyone enabled SPEAKER.

boelle, 2020-06-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

sjasonsmith, 2020-06-21:
I had reproduced this one, so I’m going to open it back up and call it confirmed.

sjasonsmith, 2020-07-04:
I believe the fan PWM will be fixed with my open PR that upgrades the STM32 framework, although the conflict when you enable SPEAKER will probably require some timer reassignments.
https://github.com/MarlinFirmware/Marlin/pull/18496

I intend to test it against that PR, but have not gotten around to it. If somebody beats me to it, please post your testing results on that PR.

closed 2020-09-08 by github-actions[bot]