MarlinFirmware/Marlin #17526: [BUG] The heating of the bed does not work properly

android444, 2020-04-13:
When you choose the heating bed, this one does not heat up. The power bed appears for a few seconds, then disappears - over and over again

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4470196/Marlin.zip)

ellensp, 2020-04-13:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted [on this page.](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question)) For best results getting help with configuration and troubleshooting, please use the following resources:

MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: [https://discord.gg/n5NJ59y](https://discord.gg/n5NJ59y)
After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

swilkens, 2020-04-13:
I see you have enabled `PIDTEMPBED` - you can try without bed PID to check if the bed is working properly.

In `Configuration.h` change `#define PIDTEMPBED` to `//#define PIDTEMPBED`

Given your fairy extreme bed PID values, I think the bed pid tuning took too long to turn up workable values. Checking with bed pid disabled is a good first check.

closed 2020-04-13 by thinkyhead

android444, 2020-04-14:
ok