MarlinFirmware/Marlin #17528: Z_Max triggered when connecting Z_Min NC switch + Sensorless homing to Z_Max

naps1saps, 2020-04-13:
I am still investigating this but have spent several hours trying to figure it out.  The only way around this that I can see is to change from a normally closed switch to a normally open switch since I can't invert Z_Max because of sensorless homing to Z_Max

Settings:
1. Sensorless homing enabled.
2. Connect normally closed switch to Z_Min (requires Z_Min inversion)
3. Enable Z_Max endstop
4. Z_Max homing_dir = 1 (home to Z_Max)
5. Pin for Z_Min when homing_dir > 0 set to Z_Min pin

Issue:
Connecting the normally closed switch to Z_Min shows Z_Max triggered.  Unplugging from Z_Min, Z_Max now shows open.

Latest stable Marlin
SKR 1.3
TMC2209
CoreXY

ellensp, 2020-04-14:
You cannot do Sensor-less homing to Z_Max on a SKR 1.3. Its electrically impossible. (without board modifications)

The instant you setup sensor-less, the firmware maps what ever you using min or max to the MIN plug pin, as you can only use the min z-min pin.

Since you have both Z-min and Z-max sensor less the firmware had no option but to swap the pins over so the Z-Min plug is really Z-MAX (sesor-less) and the Z-Max plug is really Z-MIN, 

So plug your Z-min switch in the Z-max plug. 

This is not a bug but a limitation of the controller board.

naps1saps, 2020-04-14:
I figured this out last night sort of and came back to say it was a board issue. Very strange behavior even with an NO switch being open and plugged into Z_MIN the sensorless homing doesn't work at all.  It's kind of lame the pins are still being used even with sensorless homing.  I thought all that ran through the serial communication on these boards and freed up the pins for other uses.

Overall I'm going to use a bed sensor and home to Z_MIN then have to figure out how many mm to move the bed at the end of a print.

I'd like to request a gcode feature to move to axis_max.  G162: Home axes to maximum on makerbot. and also to be able to use max and min endstops to calculate movement range/software endstops.  Seems like a simple thing yet is not implemented.

closed 2020-04-14 by naps1saps