MarlinFirmware/Marlin #17530: [BUG] SENSORLESS_PROBING not working when SENSORLESS_HOMING is disabled

tampe125, 2020-04-13:
### Bug Description

I'd wish to use SENSORLESS_PROBING without the need to use SENSORLESS_HOMING. I'm using a CoreXY setup and I'm not 100% confident sensorless homing is reliable.

SENSORLESS_PROBING: ON
SENSORELESS_HOMING: ON
I can calibrate on Z axis

SENSORLESS_PROBING: ON
SENSORELESS_HOMING: OFF
Z motor keeps smashing on the min endstop

### My Configurations
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4470791/Configuration.zip)

I disabled Z_SAFE_HOMING in SanityChecks.  
I'm using a pick and place machine, with a CoreXY setup, that's why I have endstops on the Z axes

### Steps to Reproduce

1. Compile with SENSORLESS_PROBING enabled and SENSORLESS_HOMING disabled
2. Calibrate Z

**Expected behavior:** Motor should stop when hitting the end of the Z axis

**Actual behavior:** Motors keep spinning

#### Additional Information

I already tested the bugfix branch without success

tampe125, 2020-05-19:
Any updates on this?

boelle, 2020-06-20:
@tampe125 still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle