MarlinFirmware/Marlin #17532: [BUG] IDEX sensorless homing issue with TMC2209

Zuru1, 2020-04-13:
### Bug Description

I have an IDEX 3D printer with SKR PRO V1.1 running TMC2209 in all drivers and Marlin bugfix 2.0.x, When homing the printer, X2 is homed first to X max position and doesn't get triggered which leads to the motor grinding until it's halted by the firmware. 
When the other axis  (X1 and Y) which are also using sensorless homing, they are homed to their respective min positions and function as expected.

I have also tried the following to see if the issue can be resolved to no success:

- Homing with/without QUICK_HOME
- Different stall sensitivities for X2
- With/without IMPROVE_HOMING_RELIABILITY

### My Configurations

[Config-files.zip](https://github.com/MarlinFirmware/Marlin/files/4470882/Config-files.zip)

### Steps to Reproduce
1. [First Step]
Move X1 and X2 from their home positions or to the centre of the X axis.
2. [Second Step]
Execute G28 X or T1 followed by G28 X

**Expected behavior:** [What you expect to happen]
X2 homes to the max position of X without motor grinding and stops once the stall is triggered.

**Actual behavior:** [What actually happens]
X2 moves to its home position and keeps grinding into the mount without triggering until the motor is halted by the firmware. 

Thank you for any help in advance.

Zuru1, 2020-05-02:
I think I might have found a solution/work around to this issue. The root cause is that X_MAX_PIN is assigned the diag pin for E0 while X2 is connected to E2 port. 

Have I set up wrong? Should the X2 motor connection be connected to the port after Z motor connection if IDEX setup is to be used?

Anyhow to solve the issue, I had to assign the diag pin of E2 which is connected to X2 motor to X_MAX_PIN and can now home without motor grinding into the mount.

Due to the board design, the pin definitions need to be looked over and accommodated the various setup scenarios especially when it comes to smart drivers such as TMC drivers.

Even when using mechanical endstops, I run into the same issue until I checked which pins where used for X-axis endstop pins and discovered this work around.

boelle, 2020-06-20:
@Zuru1 still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle

Zuru1, 2020-06-30:
Hi,

Sorry for the late reply, was unable to responde due to some lock down,
Yes, this is still an issue even though I'm currently running it via the hack/work around.