MarlinFirmware/Marlin #17533: STM32F4 PWM Frequency

forthereg, 2020-04-13:
So Im upgrading from one of the older 2.0x marlins on a SKR pro, which is a stm32f407zgt6. LQFP144.
i dont know what frequency it used to have but the fans were not audible.

with the new marlin all my pins are pulse width modulated at 1khz no matter the #FAN_SOFT_PWM.
I have the fans connected to mosfets running from GPIO extensions. (#define FAST_PWM_FAN is not supported on this platform btw...or so it tells me ) .

So i bit the bullet and actually assign the extention GPIOs as Fans in Pins.h and used M106 instead of M42.

but now : 
with no FAN_SOFT_PWM or scaled at 0 it runs at the same 1KHZ 

with Fan-soft-pwm  defined and scaled i get the following frequencies on fan outputs.
scale 1  15hz
scale 3  125hz
scale 4  200hz
scale 5  250hz
scale 7 is just off at S1 and 100% on at S2 .
i just like to add that the duty cycle is spot on as far as the oscilloscope shows.

this issue happened on the Main stable marlin 2.0.5.  & the nightly build and also on 2.1.
so how do i get a 20khz+ signal out of a GPIO or a fan output on the new marlin ?

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/4471650/Configuration_adv.zip)

boelle, 2020-06-20:
@forthereg is the issue still there?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle