MarlinFirmware/Marlin #17535: G38 not stopping on contact with Z probe

MxFrs, 2020-04-14:
### Bug Description

G38 is not working for me. I have a CNC and wnat to use a edgefinding touchprobe. unfortunately G38.2 does not stop on contact (even when connecting the Z Endstop contacts direcly by hand while CNC is moving). It can not be a hardware issue because G28 works perfectly using the same probing hardware. 

### My Configurations

Attatched are my configuration files and the pins file for my SKR 1.4. I had to slightly modify it to get dual endstops on Y axis to work. 
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/4472338/Config.zip)
 
If any further information regarding my hardware setup is needed feel free to ask.

MxFrs, 2020-04-16:
I found the issue for my case. In pins.h, the Z_MIN_PROBE_PIN got set to -1. 

I hardcoded my Pin in there, but I guess thats not the correct way to do it. 
`//
// Disable unused endstop / probe pins
//
#if !HAS_CUSTOM_PROBE_PIN
  #undef Z_MIN_PROBE_PIN
  #define Z_MIN_PROBE_PIN    P1_27                //changed, was -1, G38 not working with -1
#endif`

I don't know if I should close this issue even though it's not fixed but the problem was found.

thinkyhead, 2020-04-17:
You can define `Z_MIN_PROBE_PIN` in `Configuration.h` to override the default.

closed 2020-04-17 by thinkyhead