MarlinFirmware/Marlin #17539: setting up custom keypad that uses single binary output from shift register

jnour91, 2020-04-14:
### Description
Hi everyone, 
I have a 9 button keypad that has a shift register to output what button is pressed in a single line binary output, I have assigned BT1..BT2..BT3 etc to each string of binary, now I am trying to implement this into Marlin firmware, I cant seem to find the best plan of implementing, would I have to associate for example BTN_UP to BT8 (my up key)??

has anyone done this before? what would need to be changed in the firmware?

Thank you

jnour91, 2020-04-21:
Would anyone have any ideas on this, back onto to this stage again, I'm trying to find a way I can link my keypad to the marlin defined button eg "BNT_UP" which would normally take a pin input.
My action plan at this stage is to try and assign a byte eg "0b00000100 = button up" configured to BTN_UP instead of pin input.

would appreciate some of your ideas on this one.

closed 2020-05-26 by jnour91

thinkyhead, 2020-07-08:
With respect to the latest codebase…

You can simulate buttons in `MarlinUI::update_buttons` by setting the appropriate bits in `newbutton` or by setting `encoderDiff` to simulate encoder left/right rotate events. If your buttons need to be read outside of interrupt context because there's a loop involved, logic can be added to `MarlinUI::read_slow_buttons` instead.