MarlinFirmware/Marlin #17542: [BUG] Can't compile Marlin Bugfix 2.0.x for Bigtreetech SKR Mini E3

HeadlessHamsterr, 2020-04-14:
I use the default config for the skr mini e3 v1.2, but I just get errors.

[Marlin configs.zip](https://github.com/MarlinFirmware/Marlin/files/4475130/Marlin.configs.zip)

VScode output:

`In file included from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:40:0,
                 from Marlin\src\HAL\HAL_STM32F1\HAL.cpp:30:
Marlin\src\HAL\HAL_STM32F1\../../inc/SanityCheck.h:788:6: error: #error "FILAMENT_CHANGE_UNLOAD_LENGTH must be less than or equal to EXTRUDE_MAXLENGTH."
     #error "FILAMENT_CHANGE_UNLOAD_LENGTH must be less than or equal to EXTRUDE_MAXLENGTH."
      ^~~~~
In file included from Marlin\src\HAL\HAL_STM32F1\HAL.h:39:0,
                 from Marlin\src\HAL\HAL_STM32F1\HAL.cpp:31:
Marlin\src\HAL\HAL_STM32F1\timers.h: In function 'void HAL_timer_set_compare(uint8_t, hal_timer_t)':
Marlin\src\HAL\HAL_STM32F1\timers.h:147:26: error: redefinition of 'void HAL_timer_set_compare(uint8_t, hal_timer_t)'
 FORCE_INLINE static void HAL_timer_set_compare(const uint8_t timer_num, const hal_timer_t compare) {
                          ^~~~~~~~~~~~~~~~~~~~~
In file included from Marlin\src\HAL\HAL_STM32F1\../../inc/../HAL/./STM32F1/HAL.h:39:0,
                 from Marlin\src\HAL\HAL_STM32F1\../../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\HAL_STM32F1\HAL.cpp:30:
Marlin\src\HAL\HAL_STM32F1\../../inc/../HAL/./STM32F1/timers.h:147:26: note: 'void HAL_timer_set_compare(uint8_t, hal_timer_t)' previously defined here
 FORCE_INLINE static void HAL_timer_set_compare(const uint8_t timer_num, const hal_timer_t compare) {
                          ^~~~~~~~~~~~~~~~~~~~~
In file included from Marlin\src\HAL\HAL_STM32F1\HAL.h:39:0,
                 from Marlin\src\HAL\HAL_STM32F1\HAL.cpp:31:
Marlin\src\HAL\HAL_STM32F1\timers.h: In function 'void HAL_timer_isr_prologue(uint8_t)':
Marlin\src\HAL\HAL_STM32F1\timers.h:161:26: error: redefinition of 'void HAL_timer_isr_prologue(uint8_t)'
 FORCE_INLINE static void HAL_timer_isr_prologue(const uint8_t timer_num) {
                          ^~~~~~~~~~~~~~~~~~~~~~
In file included from Marlin\src\HAL\HAL_STM32F1\../../inc/../HAL/./STM32F1/HAL.h:39:0,
                 from Marlin\src\HAL\HAL_STM32F1\../../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\HAL_STM32F1\HAL.cpp:30:
Marlin\src\HAL\HAL_STM32F1\../../inc/../HAL/./STM32F1/timers.h:161:26: note: 'void HAL_timer_isr_prologue(uint8_t)' previously defined here
 FORCE_INLINE static void HAL_timer_isr_prologue(const uint8_t timer_num) {
                          ^~~~~~~~~~~~~~~~~~~~~~
In file included from Marlin\src\HAL\HAL_STM32F1\HAL.h:39:0,
                 from Marlin\src\HAL\HAL_STM32F1\HAL.cpp:31:
Marlin\src\HAL\HAL_STM32F1\timers.h: In function 'void timer_no_ARR_preload_ARPE(timer_dev*)':
Marlin\src\HAL\HAL_STM32F1\timers.h:178:26: error: redefinition of 'void timer_no_ARR_preload_ARPE(timer_dev*)'
 FORCE_INLINE static void timer_no_ARR_preload_ARPE(timer_dev *dev) {
                          ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from Marlin\src\HAL\HAL_STM32F1\../../inc/../HAL/./STM32F1/HAL.h:39:0,
                 from Marlin\src\HAL\HAL_STM32F1\../../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\HAL_STM32F1\HAL.cpp:30:
Marlin\src\HAL\HAL_STM32F1\../../inc/../HAL/./STM32F1/timers.h:178:26: note: 'void timer_no_ARR_preload_ARPE(timer_dev*)' previously defined here
 FORCE_INLINE static void timer_no_ARR_preload_ARPE(timer_dev *dev) {
                          ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from Marlin\src\HAL\HAL_STM32F1\HAL.cpp:31:0:
Marlin\src\HAL\HAL_STM32F1\HAL.h: In function 'int freeMemory()':
Marlin\src\HAL\HAL_STM32F1\HAL.h:243:12: error: redefinition of 'int freeMemory()'
 static int freeMemory() {
            ^~~~~~~~~~
In file included from Marlin\src\HAL\HAL_STM32F1\../../inc/../HAL/HAL.h:26:0,
                 from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\HAL_STM32F1\HAL.cpp:30:
Marlin\src\HAL\HAL_STM32F1\../../inc/../HAL/./STM32F1/HAL.h:244:12: note: 'int freeMemory()' previously defined here
 static int freeMemory() {
            ^~~~~~~~~~
In file included from Marlin\src\HAL\HAL_STM32F1\dogm\../../../inc/MarlinConfig.h:40:0,
                 from Marlin\src\HAL\HAL_STM32F1\dogm\u8g_com_stm32duino_fsmc.cpp:29:
Marlin\src\HAL\HAL_STM32F1\dogm\../../../inc/SanityCheck.h:788:6: error: #error "FILAMENT_CHANGE_UNLOAD_LENGTH must be less than or equal to EXTRUDE_MAXLENGTH."
     #error "FILAMENT_CHANGE_UNLOAD_LENGTH must be less than or equal to EXTRUDE_MAXLENGTH."
      ^~~~~
In file included from Marlin\src\HAL\HAL_STM32F1\dogm\../../../inc/MarlinConfig.h:40:0,
                 from Marlin\src\HAL\HAL_STM32F1\dogm\u8g_com_stm32duino_swspi.cpp:21:
Compiling .pio\build\STM32F103RC_bigtree\src\src\HAL\HAL_STM32F1\watchdog.cpp.o
Marlin\src\HAL\HAL_STM32F1\dogm\../../../inc/SanityCheck.h:788:6: error: #error "FILAMENT_CHANGE_UNLOAD_LENGTH must be less than or equal to EXTRUDE_MAXLENGTH."
[... 125 lines cut ...]
                          ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from Marlin\src\HAL\HAL_STM32F1\timers.cpp:30:0:
Marlin\src\HAL\HAL_STM32F1\HAL.h: In function 'int freeMemory()':
Marlin\src\HAL\HAL_STM32F1\HAL.h:243:12: error: redefinition of 'int freeMemory()'
 static int freeMemory() {
            ^~~~~~~~~~
In file included from Marlin\src\HAL\HAL_STM32F1\../../inc/../HAL/HAL.h:26:0,
                 from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\HAL_STM32F1\timers.cpp:29:
Marlin\src\HAL\HAL_STM32F1\../../inc/../HAL/./STM32F1/HAL.h:244:12: note: 'int freeMemory()' previously defined here
 static int freeMemory() {
            ^~~~~~~~~~
*** [.pio\build\STM32F103RC_bigtree\src\src\HAL\HAL_STM32F1\sdio.cpp.o] Error 1
*** [.pio\build\STM32F103RC_bigtree\src\src\HAL\HAL_STM32F1\timers.cpp.o] Error 1
In file included from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:40:0,
                 from Marlin\src\HAL\HAL_STM32F1\watchdog.cpp:29:
Marlin\src\HAL\HAL_STM32F1\../../inc/SanityCheck.h:788:6: error: #error "FILAMENT_CHANGE_UNLOAD_LENGTH must be less than or equal to EXTRUDE_MAXLENGTH."
     #error "FILAMENT_CHANGE_UNLOAD_LENGTH must be less than or equal to EXTRUDE_MAXLENGTH."
      ^~~~~
*** [.pio\build\STM32F103RC_bigtree\src\src\HAL\HAL_STM32F1\watchdog.cpp.o] Error 1`

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-04-14:
What part of "FILAMENT_CHANGE_UNLOAD_LENGTH must be less than or equal to EXTRUDE_MAXLENGTH" can you not understand?

Can't extract your zip file for some reason. 

The defaults in  https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/BigTreeTech/SKR%20Mini%20E3%201.2 
are #define EXTRUDE_MAXLENGTH 600 
and #define FILAMENT_CHANGE_UNLOAD_LENGTH      400
which satisfy this condition.

HeadlessHamsterr, 2020-04-14:
> What part of "FILAMENT_CHANGE_UNLOAD_LENGTH must be less than or equal to EXTRUDE_MAXLENGTH" can you not understand?
> 
Oh, missed that part. But there are still a lot of other things which I haven't changed a thing about that don't work. 

Does this zip work?
[Marlin configs.zip](https://github.com/MarlinFirmware/Marlin/files/4475344/Marlin.configs.zip)

qwewer0, 2020-04-14:
Did you try recompile it?

ellensp, 2020-04-14:
Yes that zip works.

HeadlessHamsterr, 2020-04-14:
> Did you try recompile it? (after the changes)

Yes, the error about the filament change is gone, but the redefinition errors and all the errors in the HAL directory are still there.

qwewer0, 2020-04-14:
In platformio.ini shouldn't you use STM32F103RC_btt / STM32F103RC_btt_512K instead of STM32F103RC_bigtree?

ellensp, 2020-04-14:
included platformio.ini is very old, pre name shortening...
included Configuration_adv.h is 020004
included Configuration.h is is 020005
The fact that you can compile at all indicates you using a old marlin that will accept 20004 configs.
It also shows you probably just dropped latest Configuration.h files into an old marlin.
This wont work.

qwewer0, 2020-04-14:
[Guide]( https://www.reddit.com/r/ender3/comments/e894j7/marlin_20x_guide_for_ender_3_using_skr_mini_e3_v12/ ) for the board.

qwewer0, 2020-04-14:
I think this issue can be closed.

ellensp, 2020-04-14:
I did find one issue. after I upgraded to current bugfix 
They enable LCD_BED_LEVELING which enables MESH_EDIT_MENU
resulting in a actual bug  
```
Marlin/src/lcd/menu/menu_bed_leveling.cpp: In function 'void menu_edit_mesh()':
Marlin/src/lcd/menu/menu_bed_leveling.cpp:214:47: error: 'Z_VALUES' was not declared in this scope
```
The cause in bedlevel.h is HAS_MESH and only defines Z_VALUES if the test passes.
The test HAS_MESH is #if ANY(AUTO_BED_LEVELING_BILINEAR, AUTO_BED_LEVELING_UBL, MESH_BED_LEVELING)
And non of these are defined.. they have AUTO_BED_LEVELING_LINEAR

I don't know if this is an invalid combination (in which case it conditionals should catch it) or the test should include AUTO_BED_LEVELING_LINEAR...

HeadlessHamsterr, 2020-04-14:
> included platformio.ini is very old, pre name shortening...
> included Configuration_adv.h is 020004
> included Configuration.h is is 020005
> The fact that you can compile at all indicates you using a [...]

Jesus, I'm an idiot. A lot of copying and pasting config files was bound to go wrong I guess. Thanks!

closed 2020-04-14 by HeadlessHamsterr