MarlinFirmware/Marlin #17545: [BUG] (short description)

HeadlessHamsterr, 2020-04-14:
I downloaded the latest Marlin and modified the config and advanced config according to this guide: https://www.reddit.com/r/ender3/comments/e894j7/marlin_20x_guide_for_ender_3_using_skr_mini_e3_v12/

When I try to compile with VScode I get this message:

`.pio/build/STM32F103RC_btt_512K/src/src/HAL/STM32F1/persistent_store_flash.cpp.o: In function `PersistentStore::access_start()':
C:\Users\joost\Desktop\Main Marlin/Marlin\src\HAL\STM32F1/persistent_store_flash.cpp:50: multiple definition of `PersistentStore::access_start()'
.pio/build/STM32F103RC_btt_512K/src/src/HAL/STM32F1/eeprom_flash.cpp.o:C:\Users\joost\Desktop\Main Marlin/Marlin\src\HAL\STM32F1/eeprom_flash.cpp:49: first defined here
.pio/build/STM32F103RC_btt_512K/src/src/HAL/STM32F1/persistent_store_flash.cpp.o: In function `PersistentStore::access_finish()':
C:\Users\joost\Desktop\Main Marlin/Marlin\src\HAL\STM32F1/persistent_store_flash.cpp:63: multiple definition of `PersistentStore::access_finish()'
.pio/build/STM32F103RC_btt_512K/src/src/HAL/STM32F1/eeprom_flash.cpp.o:C:\Users\joost\Desktop\Main Marlin/Marlin\src\HAL\STM32F1/eeprom_flash.cpp:62: first defined here
.pio/build/STM32F103RC_btt_512K/src/src/HAL/STM32F1/persistent_store_flash.cpp.o: In function `PersistentStore::write_data(int&, unsigned char const*, unsigned int, unsigned short*)':
C:\Users\joost\Desktop\Main Marlin/Marlin\src\HAL\STM32F1/persistent_store_flash.cpp:93: multiple definition of `PersistentStore::write_data(int&, unsigned char const*, unsigned int, unsigned short*)'
.pio/build/STM32F103RC_btt_512K/src/src/HAL/STM32F1/eeprom_flash.cpp.o:C:\Users\joost\Desktop\Main Marlin/Marlin\src\HAL\STM32F1/eeprom_flash.cpp:92: first defined here
.pio/build/STM32F103RC_btt_512K/src/src/HAL/STM32F1/persistent_store_flash.cpp.o: In function `PersistentStore::read_data(int&, unsigned char*, unsigned int, unsigned short*, bool)':
C:\Users\joost\Desktop\Main Marlin/Marlin\src\HAL\STM32F1/persistent_store_flash.cpp:101: multiple definition of `PersistentStore::read_data(int&, unsigned char*, unsigned int, unsigned short*, bool)'
.pio/build/STM32F103RC_btt_512K/src/src/HAL/STM32F1/eeprom_flash.cpp.o:C:\Users\joost\Desktop\Main Marlin/Marlin\src\HAL\STM32F1/eeprom_flash.cpp:100: first defined here
.pio/build/STM32F103RC_btt_512K/src/src/HAL/STM32F1/persistent_store_flash.cpp.o: In function `PersistentStore::capacity()':
C:\Users\joost\Desktop\Main Marlin/Marlin\src\HAL\STM32F1/persistent_store_flash.cpp:109: multiple definition of `PersistentStore::capacity()'
.pio/build/STM32F103RC_btt_512K/src/src/HAL/STM32F1/eeprom_flash.cpp.o:C:\Users\joost\Desktop\Main Marlin/Marlin\src\HAL\STM32F1/eeprom_flash.cpp:108: first defined here
.pio/build/STM32F103RC_btt_512K/src/src/HAL/shared/persistent_store_api.cpp.o:C:\Users\joost\Desktop\Main Marlin/Marlin\src\HAL\shared/persistent_store_api.h:53: multiple definition of `persistentStore'
.pio/build/STM32F103RC_btt_512K/src/src/HAL/shared/eeprom_api.cpp.o:C:\Users\joost\Desktop\Main Marlin/Marlin\src\HAL\shared/eeprom_api.h:53: first defined here
.pio/build/STM32F103RC_btt_512K/src/src/gcode/sdcard/M20.cpp.o: In function `GcodeSuite::M20()':
C:\Users\joost\Desktop\Main Marlin/Marlin\src\gcode\sdcard/M20.cpp:33: multiple definition of `GcodeSuite::M20()'
.pio/build/STM32F103RC_btt_512K/src/src/gcode/sd/M20.cpp.o:C:\Users\joost\Desktop\Main Marlin/Marlin\src\gcode\sd/M20.cpp:33: first defined here
.pio/build/STM32F103RC_btt_512K/src/src/gcode/sdcard/M21_M22.cpp.o: In function `GcodeSuite::M21()':
C:\Users\joost\Desktop\Main Marlin/Marlin\src\gcode\sdcard/M21_M22.cpp:33: multiple definition of `GcodeSuite::M21()'
.pio/build/STM32F103RC_btt_512K/src/src/gcode/sd/M21_M22.cpp.o:C:\Users\joost\Desktop\Main Marlin/Marlin\src\gcode\sd/M21_M22.cpp:33: first defined here
.pio/build/STM32F103RC_btt_512K/src/src/gcode/sdcard/M21_M22.cpp.o: In function `GcodeSuite::M22()':
C:\Users\joost\Desktop\Main Marlin/Marlin\src\gcode\sdcard/M21_M22.cpp:38: multiple definition of `GcodeSuite::M22()'
.pio/build/STM32F103RC_btt_512K/src/src/gcode/sd/M21_M22.cpp.o:C:\Users\joost\Desktop\Main Marlin/Marlin\src\gcode\sd/M21_M22.cpp:38: first defined here
.pio/build/STM32F103RC_btt_512K/src/src/gcode/sdcard/M23.cpp.o: In function `GcodeSuite::M23()':
C:\Users\joost\Desktop\Main Marlin/Marlin\src\gcode\sdcard/M23.cpp:38: multiple definition of `GcodeSuite::M23()'
.pio/build/STM32F103RC_btt_512K/src/src/gcode/sd/M23.cpp.o:C:\Users\joost\Desktop\Main Marlin/Marlin\src\gcode\sd/M23.cpp:38: first defined here
.pio/build/STM32F103RC_btt_512K/src/src/gcode/sdcard/M24_M25.cpp.o: In function `GcodeSuite::M24()':
C:\Users\joost\Desktop\Main Marlin/Marlin\src\gcode\sdcard/M24_M25.cpp:50: multiple definition of `GcodeSuite::M24()'
.pio/build/STM32F103RC_btt_512K/src/src/gcode/sd/M24_M25.cpp.o:C:\Users\joost\Desktop\Main Marlin/Marlin\src\gcode\sd/M24_M25.cpp:50: first defined here
.pio/build/STM32F103RC_btt_512K/src/src/gcode/sdcard/M24_M25.cpp.o: In function `GcodeSuite::M25()':
C:\Users\joost\Desktop\Main Marlin/Marlin\src\gcode\sdcard/M24_M25.cpp:87: multiple definition of `GcodeSuite::M25()'
.pio/build/STM32F103RC_btt_512K/src/src/gcode/sd/M24_M25.cpp.o:C:\Users\joost\Desktop\Main Marlin/Marlin\src\gcode\sd/M24_M25.cpp:87: first defined here
.pio/build/STM32F103RC_btt_512K/src/src/gcode/sdcard/M26.cpp.o: In function `GcodeSuite::M26()':
C:\Users\joost\Desktop\Main Marlin/Marlin\src\gcode\sdcard/M26.cpp:33: multiple definition of `GcodeSuite::M26()'
.pio/build/STM32F103RC_btt_512K/src/src/gcode/sd/M26.cpp.o:C:\Users\joost\Desktop\Main Marlin/Marlin\src\gcode\sd/M26.cpp:33: first defined here
.pio/build/STM32F103RC_btt_512K/src/src/gcode/sdcard/M27.cpp.o: In function `GcodeSuite::M27()':
C:\Users\joost\Desktop\Main Marlin/Marlin\src\gcode\sdcard/M27.cpp:35: multiple definition of `GcodeSuite::M27()'
.pio/build/STM32F103RC_btt_512K/src/src/gcode/sd/M27.cpp.o:C:\Users\joost\Desktop\Main Marlin/Marlin\src\gcode\sd/M27.cpp:35: first defined here
.pio/build/STM32F103RC_btt_512K/src/src/gcode/sdcard/M28_M29.cpp.o: In function `GcodeSuite::M28()':
C:\Users\joost\Desktop\Main Marlin/Marlin\src\gcode\sdcard/M28_M29.cpp:61: multiple definition of `GcodeSuite::M28()'
.pio/build/STM32F103RC_btt_512K/src/src/gcode/sd/M28_M29.cpp.o:C:\Users\joost\Desktop\Main Marlin/Marlin\src\gcode\sd/M28_M29.cpp:61: first defined here
.pio/build/STM32F103RC_btt_512K/src/src/gcode/sdcard/M28_M29.cpp.o: In function `GcodeSuite::M29()':
C:\Users\joost\Desktop\Main Marlin/Marlin\src\gcode\sdcard/M28_M29.cpp:71: multiple definition of `GcodeSuite::M29()'
.pio/build/STM32F103RC_btt_512K/src/src/gcode/sd/M28_M29.cpp.o:C:\Users\joost\Desktop\Main Marlin/Marlin\src\gcode\sd/M28_M29.cpp:71: first defined here
.pio/build/STM32F103RC_btt_512K/src/src/gcode/sdcard/M30.cpp.o: In function `GcodeSuite::M30()':
C:\Users\joost\Desktop\Main Marlin/Marlin\src\gcode\sdcard/M30.cpp:33: multiple definition of `GcodeSuite::M30()'
.pio/build/STM32F103RC_btt_512K/src/src/gcode/sd/M30.cpp.o:C:\Users\joost\Desktop\Main Marlin/Marlin\src\gcode\sd/M30.cpp:33: first defined here
.pio/build/STM32F103RC_btt_512K/src/src/gcode/sdcard/M32.cpp.o: In function `GcodeSuite::M32()':
C:\Users\joost\Desktop\Main Marlin/Marlin\src\gcode\sdcard/M32.cpp:43: multiple definition of `GcodeSuite::M32()'
.pio/build/STM32F103RC_btt_512K/src/src/gcode/sd/M32.cpp.o:C:\Users\joost\Desktop\Main Marlin/Marlin\src\gcode\sd/M32.cpp:43: first defined here
.pio/build/STM32F103RC_btt_512K/src/src/gcode/sdcard/M524.cpp.o: In function `GcodeSuite::M524()':
C:\Users\joost\Desktop\Main Marlin/Marlin\src\gcode\sdcard/M524.cpp:35: multiple definition of `GcodeSuite::M524()'
.pio/build/STM32F103RC_btt_512K/src/src/gcode/sd/M524.cpp.o:C:\Users\joost\Desktop\Main Marlin/Marlin\src\gcode\sd/M524.cpp:35: first defined here
.pio/build/STM32F103RC_btt_512K/src/src/gcode/sdcard/M928.cpp.o: In function `GcodeSuite::M928()':
C:\Users\joost\Desktop\Main Marlin/Marlin\src\gcode\sdcard/M928.cpp:35: multiple definition of `GcodeSuite::M928()'
.pio/build/STM32F103RC_btt_512K/src/src/gcode/sd/M928.cpp.o:C:\Users\joost\Desktop\Main Marlin/Marlin\src\gcode\sd/M928.cpp:35: first defined here
.pio/build/STM32F103RC_btt_512K/src/src/gcode/temperature/M104_M109.cpp.o: In function `GcodeSuite::M104()':
C:\Users\joost\Desktop\Main Marlin/Marlin\src\gcode\temperature\../parser.h:141: multiple definition of `GcodeSuite::M104()'
[... cut ...]
.pio/build/STM32F103RC_btt_512K/src/src/lcd/extui/lib/ftdi_eve_touch_ui/language/language.cpp.o:C:\Users\joost\Desktop\Main Marlin/Marlin\src\lcd\extui\lib\ftdi_eve_touch_ui\language/language.cpp:27: multiple definition of `lang'
.pio/build/STM32F103RC_btt_512K/src/src/lcd/extensible_ui/lib/ftdi_eve_touch_ui/language/language.cpp.o:C:\Users\joost\Desktop\Main Marlin/Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\language/language.cpp:27: first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\STM32F103RC_btt_512K\firmware.elf] Error 1`

[Marlin configs.zip](https://github.com/MarlinFirmware/Marlin/files/4477084/Marlin.configs.zip)

dkggpeters, 2020-04-14:
Did you start with the Bigtree or Ender 3 default config files?

dkggpeters, 2020-04-14:
Comment out line 35 SailfishLCD=https://github.com/mikeshub/SailfishLCD/archive/master.zip in  platformio.ini

qwewer0, 2020-04-14:
I don't get any error with your configuration.

HeadlessHamsterr, 2020-04-14:
> Did you start with the Bigtree or Ender 3 default config files?

I started with the default marlin config

HeadlessHamsterr, 2020-04-14:
> I don't get any error with your configuration files.

Hmm. Maybe it's my pc/vscode?

dkggpeters, 2020-04-14:
It compiles for me as well with no errors.

qwewer0, 2020-04-14:
@HeadlessHamsterr Try to reinstall platformio, then VSCode if it doesn't help.

HeadlessHamsterr, 2020-04-14:
> It compiles for me as well with no errors.

Weird

HeadlessHamsterr, 2020-04-14:
> @HeadlessHamsterr Try to reinstall platformio, then VSCode if it doesn't help.

Will do

HeadlessHamsterr, 2020-04-14:
Reinstall didn't work, had this problem before and a windows reinstall did work but I don't feel like doing that haha. Tried my laptop, which does work so I guess I'll use that one. I'll close this issue unless someone has an other idea?

GMagician, 2020-04-15:
Try to delete .pio folder and compile again

HeadlessHamsterr, 2020-04-15:
Nope didn't do it unfortunatly

GMagician, 2020-04-15:
Then delete all marlin folder (but your config), re-download it from github, and restore your config...

closed 2020-04-16 by thinkyhead