MarlinFirmware/Marlin #17546: controllerFan' was not declared in power.cpp

Cyberzoid1, 2020-04-14:
### Bug Description

Compiling with AUTO_POWER_CONTROL and USE_CONTROLLER_FAN fails to compile due to `controllerFan not declared in scope`

### My Configurations
Relevant Configuration.h
```C++
#define PSU_CONTROL
//#define PSU_NAME "Power Supply"

#if ENABLED(PSU_CONTROL)
  #define PSU_ACTIVE_HIGH false     // Set 'false' for ATX, 'true' for X-Box

  //#define PSU_DEFAULT_OFF         // Keep power off until enabled directly with M80
  //#define PSU_POWERUP_DELAY 100   // (ms) Delay for the PSU to warm up to full power

  #define AUTO_POWER_CONTROL      // Enable automatic control of the PS_ON pin
  #if ENABLED(AUTO_POWER_CONTROL)
    #define AUTO_POWER_FANS         // Turn on PSU if fans need power
    #define AUTO_POWER_E_FANS
    #define AUTO_POWER_CONTROLLERFAN
    #define AUTO_POWER_CHAMBER_FAN
    //#define AUTO_POWER_E_TEMP        50 // (°C) Turn on PSU over this temperature
    //#define AUTO_POWER_CHAMBER_TEMP  30 // (°C) Turn on PSU over this temperature
    #define POWER_TIMEOUT 30
  #endif
#endif
```
Relevant Configuration_adv.h code from my settings
```C++
#define USE_CONTROLLER_FAN
#if ENABLED(USE_CONTROLLER_FAN)
  #define CONTROLLER_FAN_PIN 57        // Set a custom pin for the controller fan
  //#define CONTROLLER_FAN_USE_Z_ONLY    // With this option only the Z axis is considered
  #define CONTROLLERFAN_SPEED_MIN      0 // (0-255) Minimum speed. (If set below this value the fan is turned off.)
  #define CONTROLLERFAN_SPEED_ACTIVE 255 // (0-255) Active speed, used when any motor is enabled
  #define CONTROLLERFAN_SPEED_IDLE     0 // (0-255) Idle speed, used when motors are disabled
  #define CONTROLLERFAN_IDLE_TIME     60 // (seconds) Extra time to keep the fan running after disabling motors
  //#define CONTROLLER_FAN_EDITABLE      // Enable M710 configurable settings
  #if ENABLED(CONTROLLER_FAN_EDITABLE)
    #define CONTROLLER_FAN_MENU          // Enable the Controller Fan submenu
  #endif
#endif
```

### Steps to Reproduce

1. Checkout https://github.com/MarlinFirmware/Marlin/releases/tag/2.0.5.3
2. Enable PSU_Control Line   Configuration.h:322
3. Enable AUTO_POWER_CONTROL   Configuration.h:331
4. Enable USE_CONTROLLER_FAN     Configuration_adv.h:345
5. Compile

**Expected behavior:** [What you expect to happen]
Compile successful

**Actual behavior:** [What actually happens]

> Marlin\src\feature\power.cpp: In static member function 'static bool Power::is_power_needed()':
Marlin\src\feature\power.cpp:50:9: error: 'controllerFan' was not declared in this scope
     if (controllerFan.state()) return true;
         ^
*** [.pio\build\mega2560\src\src\feature\power.cpp.o] Error 1

#### Additional Information
Related Pull: https://github.com/MarlinFirmware/Marlin/pull/17149
[GH_Issue_ConrollerFan.zip](https://github.com/MarlinFirmware/Marlin/files/4477313/GH_Issue_ConrollerFan.zip)

My build environment:
Visual Studio Code with PlatformIO
Windows 10

**Potential fix:**
I added `#include "controllerfan.h"` to `power.cpp`. Compiles successfully.

thinkyhead, 2020-04-26:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

Cyberzoid1, 2020-04-26:
I just downloaded your attached bugfix-2.0.x.zip

- Build as-is successful
- Build with steps to reproduce above yields the same error
- Build with steps to reproduce & `#include "controllerfan.h"` potential fix builds successfully

boelle, 2020-06-20:
@Cyberzoid1 is the issue still there?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle

wap32, 2020-07-08:
I can confirm the issue is still there.
Just tried it with the latest bugfix-2.0.x and compilation fails as in the original description.

Adding `#include "controllerfan.h"` as suggested appears to fix the issue (meaning it compiles and behaves as expected).

thinkyhead, 2020-07-09:
Thanks for the report! The missing `#include` has now been added.