MarlinFirmware/Marlin #17551: M48 probe test, stops after homing if triggered from menu

tomarigr, 2020-04-14:
Hi, M48 probe test stops right after homing if you trigger it from menu.

no error messages on the screen or console.

doing m48 from console works just fine though

using latest stable 2.0.5.3 on skr e3 mini v1.2 + servo probe

tomarigr, 2020-04-28:
bump, no one saw this?

qwewer0, 2020-04-28:
Works for me.
SKR Mini E3 v1.2, BLTouch v3.1

tomarigr, 2020-04-28:
Doesn't work for me since marlin 2.0.4.*
SKR Mini E3 v1.2
servo probe
Psu control
Active endstops *might be this one, endstops are hit during initial homing after you select m48 from menu
Ubl
Pid temp control bed and hotend

tomarigr, 2020-05-02:
Video of what happens.
Basically it just homes. 

Could rename the menu item to "just home" and leave it as is!

https://youtu.be/yyJ3IRuTyW0

boelle, 2020-06-20:
@tomarigr still an issue?

tomarigr, 2020-06-21:
Hi, has anything changed from the post date? As of about a month ago yes, it is still an issue. Haven't checked the bugfix version though. Will do at first chance and report

tomarigr, 2020-06-28:
@boelle just checked latest bugfix and the problem still exists. It just homes and that's it.

sjasonsmith, 2020-07-04:
@tomarigr, can you attach your configuration files as required in the issue template? That will make it easier for somebody to try to reproduce the problem you are experiencing.

thinkyhead, 2020-07-16:
The `M48` command will bail if it detects no homing. But otherwise, it will proceed. Either the `G28` command is somehow trouncing the `M48` command before it ends up in the buffer, or the `M48` command is bailing because it thinks homing failed. But, there are some things you can do to figure out which one.

First, grab the latest `bugfix-2.0.x` branch. Tweak the following menu item in `menu_motion.cpp`… and add a `V4` to the `M48` command so it gives extra output:
```cpp
GCODES_ITEM(MSG_M48_TEST, PSTR("G28\nM48 P10"));
```

After flashing the new firmware, connect from your favorite host and turn on extra logging with `M111 S3`. Finally, use your LCD controller to do the Repeatability Test. The log output should tell us pretty quickly whether `M48` is getting executed at all.

closed 2020-08-21 by github-actions[bot]