MarlinFirmware/Marlin #17553: [BUG] G33 delta auto calibration does not converge to the proper calibration values

jafidod, 2020-04-15:
It looks like that the saved/default hard coded values are not used for G33 calibration or the calibration weight functions are not properly configured. The height is measured properly but in case of delta tower, endstop and radius calibration, it looks like that center has offset to Z tower and radius is measured much lower than the real one. I've measured physical parameters according to the following picture:
[image]
and I am able to reach each position of 240mm bed with max 1 mm error in Z.
### My Configurations
(https://github.com/MarlinFirmware/Marlin/files/4477049/Configuration.zip)

### Steps to Reproduce

1. Modify Anycubic Kossel Linear Plus
2. Run G33 autocalibration with parameter P>1
3. Observe navigation to the strange positions mainly around Z tower

#### Additional Information
https://youtu.be/XSB2MxVS_dY

jafidod, 2020-04-18:
It looks like that the board skips steps which is weird because it has enough performance to compute basic action like calibration. I've swap motor, drivers, checked geometry to exclude mechanical issues but the problem persist. Of course the board itself can be broken but I it is strange that it behaves all the time in the same way.

jafidod, 2020-04-19:
I've tried Klipper today and the calibration there works as expected at least with TMC2209 drivers in standalone mode. There has to be something fishy in the Marlin firmware but I cannot figure out what is the root cause.

jafidod, 2020-04-20:
Results from Klipper delta calibration executed 10 times in a row. (G28 followed by the DELTA_CALIBRATION command). It looks pretty solid, so I will try to hardcode these values to Marlin to check the behavior during printing.

> stepper_a: position_endstop: 222.168284 angle: 209.584392 arm: 254.000000
> stepper_b: position_endstop: 223.105789 angle: 329.948278 arm: 254.000000
> stepper_c: position_endstop: 224.277318 angle: 90.000000 arm: 254.000000
> delta_radius: 131.575241
> 
> stepper_a: position_endstop: 222.154928 angle: 209.593461 arm: 254.000000
> stepper_b: position_endstop: 223.105024 angle: 329.948834 arm: 254.000000
> stepper_c: position_endstop: 224.254342 angle: 90.000000 arm: 254.000000
> delta_radius: 131.526960
> 
> stepper_a: position_endstop: 222.196082 angle: 209.593208 arm: 254.000000
> stepper_b: position_endstop: 223.147591 angle: 329.970015 arm: 254.000000
> stepper_c: position_endstop: 224.356562 angle: 90.000000 arm: 254.000000
> delta_radius: 131.571045
> 
> stepper_a: position_endstop: 222.318806 angle: 209.599253 arm: 254.000000
> stepper_b: position_endstop: 223.040750 angle: 329.973014 arm: 254.000000
> stepper_c: position_endstop: 224.374850 angle: 90.000000 arm: 254.000000
> delta_radius: 131.549395
> 
> stepper_a: position_endstop: 222.187311 angle: 209.602485 arm: 254.000000
> stepper_b: position_endstop: 223.168660 angle: 329.969983 arm: 254.000000
> stepper_c: position_endstop: 224.380063 angle: 90.000000 arm: 254.000000
> delta_radius: 131.505607
> 
> stepper_a: position_endstop: 222.234452 angle: 209.590782 arm: 254.000000
> stepper_b: position_endstop: 223.091381 angle: 329.980326 arm: 254.000000
> stepper_c: position_endstop: 224.372427 angle: 90.000000 arm: 254.000000
> delta_radius: 131.497491
> 
> stepper_a: position_endstop: 222.192077 angle: 209.614132 arm: 254.000000
> stepper_b: position_endstop: 223.278705 angle: 329.971793 arm: 254.000000
> stepper_c: position_endstop: 224.305711 angle: 90.000000 arm: 254.000000
> delta_radius: 131.543224
> 
> stepper_a: position_endstop: 222.033394 angle: 209.601885 arm: 254.000000
> stepper_b: position_endstop: 223.202497 angle: 330.000228 arm: 254.000000
> stepper_c: position_endstop: 224.318468 angle: 90.000000 arm: 254.000000
> delta_radius: 131.528523
> 
> stepper_a: position_endstop: 222.191186 angle: 209.597087 arm: 254.000000
> stepper_b: position_endstop: 223.238349 angle: 329.961597 arm: 254.000000
> stepper_c: position_endstop: 224.348757 angle: 90.000000 arm: 254.000000
> delta_radius: 131.527955
> 
> stepper_a: position_endstop: 222.089632 angle: 209.601999 arm: 254.000000
> stepper_b: position_endstop: 223.198400 angle: 329.985156 arm: 254.000000
> stepper_c: position_endstop: 224.342472 angle: 90.000000 arm: 254.000000
> delta_radius: 131.525220

jafidod, 2020-04-24:
My current understanding of the issue is that Gama tower connected to the stepper Z loses some steps or managing it incorrectly. The trajectory becomes unpredictable and in the end the nozzle hits the bed or it flies in the sky. Strange is that it is happening just during the G33 autocalibration or at least I am not aware of any other situation with such a behavior.
I've tried to have a look to G33.h and G33.cpp files but I am able to find there any suspicious code. I've tried to even roll back Trinamic library from version 0.6.2 to 0.5.0 without change of the behavior.
The thing is I am becoming suspicious to the board itself and the drivers. I swapped the drivers, I swapped the motors connection to just check whether it is not causing the issue but the behavior is constant.
I've remapped pins assigned to the Z to the E1 port by modification of pins_BTT_SKR_V1_3.h and pins_BTT_SKR_common.h today.
// modified E1 to become Z instead
#define Z_STEP_PIN                         P0_01
#define Z_DIR_PIN                          P0_00
#define Z_ENABLE_PIN                       P0_10
#ifndef Z_CS_PIN
#define Z_CS_PIN                         P1_01
#endif
I've expected the same behavior as I know but strange thing happen because the Gama tower motor connected to the E1 becomes much faster than the others. I haven't changed any acceleration, feed rate or settings of the Marlin, the only issued change is the port remapping.

If you have any suggestion what should I try let me know because I am out of ideas.

hex218, 2020-05-06:
I experience the same problems, also on a SKR (v1.4) and TMCs in Standalone Mode

jafidod, 2020-05-07:
I spent another day on investigating the issue and I believe that I have found workaround

1) I've tried AT mega 2560 board Trigorilla v1.1 with A4988 and the behavior didn't change. 
2) I've slowed down the probing speed to 400 and it looks that it is not loosing steps anymore.
> #define XY_PROBE_SPEED 400
> #define Z_PROBE_SPEED_FAST 400
3) The same behavior was observed with the SKR 1.3 and TMC2209 too. Slowing down the probing helps even it makes no sense. Now I am 100% sure that it is a Marlin issue and not the HW one.

boelle, 2020-06-20:
@jafidod is this still an issue?

jafidod, 2020-06-21:
Yep the issue is still present. I have no idea what is wrong in Marlin. As a workaround I have ported my config to the Reprap firmware https://www.jayuk.org/

boelle, 2020-06-29:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-29 by boelle