MarlinFirmware/Marlin #17555: Won't compile properly in VS Code due to conditionals in macro calls

rbrian, 2020-04-15:
### Bug Description

In src\module\configuration_store.cpp, there are several cases of code similar to:

    CONFIG_ECHO_HEADING("Maximum feedrates (units/s):");
    CONFIG_ECHO_START();
    SERIAL_ECHOLNPAIR_P(
        PSTR("  M203 X"), LINEAR_UNIT(planner.settings.max_feedrate_mm_s[X_AXIS])
      , SP_Y_STR, LINEAR_UNIT(planner.settings.max_feedrate_mm_s[Y_AXIS])
      , SP_Z_STR, LINEAR_UNIT(planner.settings.max_feedrate_mm_s[Z_AXIS])
      #if DISABLED(DISTINCT_E_FACTORS)
        , SP_E_STR, VOLUMETRIC_UNIT(planner.settings.max_feedrate_mm_s[E_AXIS])
      #endif
    );

which results in a non-fatal error "improperly terminated macro invocation" because of the #if inside the macro call. It seems that this code can have undocumented results. gcc accepts it, but msvc doesn't. see https://stackoverflow.com/questions/19111383/ifdef-inside-a-macro-call-works-with-gcc-but-not-with-msvc for a discussion of the topic. 

### My Configurations

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/4479087/Configs.zip)

### Steps to Reproduce

1.  open the project in vs code
2. build with platformio

**Expected behavior:** [What you expect to happen]
no "problems" encountered while building 

**Actual behavior:** [What actually happens]

Non-fatal error message "improperly terminated macro invocation"

#### Additional Information

screenshot of error message at https://www.screencast.com/t/7CxiZst4lIdr

sl1pkn07, 2020-04-15:
this project not use MSVC. how you set MSVC as compiler in platformio?

rbrian, 2020-04-16:
I didn't change anything from the default installation of platformio. All I know is that it shows it as a "problem" when building, but the build doesn't fail because of it. Maybe the build succeeds but the intellisense engine or something triggers on it?

I was just worried that when I see it, it's maybe going to wipe out a good portion of that file due to the error, so I "fixed" them all by either commenting out the #if and #endif, if the condition was true, or commenting out the whole block otherwise.

sl1pkn07, 2020-04-16:
you need just build with the menu in the left of VScode (need click first in the "bug" in the left toolbar) selecting you MCU environment (in you config uses SKR 1.4 turgo, then need set LPC1789)

[image]

if the build is failed, then need see the console

[image]
this case is all green, build OK

in your config, the error is

~~~
Compiling .pio/build/LPC1769/src/src/HAL/LPC1768/u8g/u8g_com_HAL_LPC1768_sw_spi.cpp.o
In file included from Marlin/src/HAL/LPC1768/../../core/../inc/MarlinConfig.h:40:0,
                 from Marlin/src/HAL/LPC1768/../../core/serial.h:24,
                 from Marlin/src/HAL/LPC1768/DebugMonitor.cpp:26:
Marlin/src/HAL/LPC1768/../../core/../inc/SanityCheck.h:1233:8: error: #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires the Z_MIN_PIN to be defined."
       #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires the Z_MIN_PIN to be defined."
        ^~~~~
~~~

you can go to terminal scrollbar in te right and up it to view the error

[image]

greetings

closed 2020-04-17 by thinkyhead