MarlinFirmware/Marlin #17556: [FR] G-code to reset control board (e.g. for firmware upload)

jan-gerard, 2020-04-15:
### Description
Add a G-code command that resets the printer control board.

I can flash the firmware of my printer via the ESP3D (using the OctoPrint-FirmwareUpdater plugin), but then I have to press the reset button (below the LCD) just before starting the firmware upload.
It would be great if this plugin would reset the printer just before starting the firmware upload.
However, I tried with the G997 code, but that can only be enabled in the printer for a hand full of board configurations, not for the Anet. However, I'm sure Marlin could reset the board through some native Arduino command. 
The server code or plugin could then first send this G-code, then wait for a moment, and then start uploading the firmware. Most microprocessors have a bootloader that accepts new binary code on the (serial) input just after restarting. Timing is critical, therefore a manual reset is not reliable.

### Feature Workflow
1. Marlin receives G-code
2. Program running on printer control board resets the board
3. Microcontroller bootloader accepts new firmware and restarts after loading

#### Additional Information
I use an Anet A6 (Marlin 2.0) with ESP8266 running ESP3D.

retikulumx, 2020-04-15:
For this you probably have to wire some output of the chip to the reset pin, so it can reset itself.

jan-gerard, 2020-04-15:
I read that this could do the job. Assuming the bootloader starts at mem location 0.
```
void (softReset){
   asm volatile ("  jmp 0");
}
```

retikulumx, 2020-04-15:
Ah, a soft reset. Yeah, possible if that's enough to upload a new firmware.

jan-gerard, 2020-04-15:
If there is a G-code that can be issued by the firmware uploader, that controls the serial port, and it is executed within a limited time (or it gives a response that it was succesful), then I think that it is doable. For example with an Octoprint plugin. 
1) Send G-reset
2) Wait for response (or not, if reset happens first)
3) Upload formware, since controller board is in flash mode
4) done

Powerprobot, 2020-04-15:
The latest Bugfix works with M997. My enviroment is SKR1.3 with the LPC1768 Chip. After the M997 command, the Board make a Reset and flash the BIN-File from OnBoard microSD. Mostly is Marlin define to use the SD-Card from the LCD-Display. If you only use Octoprint without Display or without SD-Slot on the LCD, then can configure Marlin to use the microSD-Slot OnBoard.

jan-gerard, 2020-04-16:
Marlin on my Anet A6 uses the onboard microSD. The firmware flashing is handled via the serial interface. Either via the CH340 chip that is able to pull the reset line of the microcontroller, or via a serial interface to a WiFi board (EPS8266) and Tx/Rx lines only. In the last case, I need to reset the microcontroller just before starting to send the new firware over the Tx/Rx interface. 
The board configuration for the Anet microcontroller is refering to Atmel hardware, and that does not provide handling of the M997 command. 
If M997 is the right command to do this, that should be implemented then. But what I read is that M997 is for firmware update via SD card. I don't intend to do that. But it's the g-code command to what is closest to what is needed.
I am surprised that there is no dedicated g-code command to reboot/reset a controller. It isn't M999 either.