MarlinFirmware/Marlin #17557: SKR 1.3 DHT Humidity & Temperature Sensor

TheCloudIdea, 2020-04-15:
hello
I modify **CUBEPRO DUE**, it have inside **CIRRUS60** heater, already can control it with bed heater on lcd menu.
I trying to instal **DHT11**-**DHT21**-**DHT22** to **skr 1.3** with screen info, but cant find step by step how to do `http://i.imgur.com/hnKqJ9P.jpg`
marlin 2.0 not have any info about it, but found in old version some info, but it to old...
any helps?

astorun, 2020-05-29:
Yes I would really like to do that too. This is I believe from this https://reprap.org/forum/read.php?146,449126

I think that guy were using Atmel MCU. I believe The DHT libraries are not compatible with LPC176x series so I could not be able to run it on my SKR 1.4 Turbo as well.

I will really appreciate if one of the maintainers can look into this. I think its nice to see DHT series sensor support for Marlin will be useful, especially if you have a drybox for filament. You can check it on printer screen right away.

ellensp, 2020-05-29:
Marlin 2 is a very different beast
I have just gotten the DHT11 working under marlin 2 on a ramps using platformio see https://reprap.org/forum/read.php?1,460828,873148#msg-873148

It uses two libraries, 
DHT sensor library 1.3.10
Adafruit Unified Sensor 1.1.3 (and you must stop it including Adafruit ADXL343, wich it wants, but isn't needed) 
Both of these are available for mega2560 but not the lpc1768/9 
(there are no arduino libraries for lpc1768/9 other than what marlin people write.)

I have both libraries compiling ok on lpc1768, but not at all tested...   
long way to go...

astorun, 2020-05-29:
I contacted to Adafruit and they said its impossible to make it work with LPC series since this library is only works for AVR MCUs.

I am really wishing to make this feature works natively. If one of the maintainers are willing to integrate this I am ready to start making monthly donations for Marlin Firmware development via github (I am serious). Since I am not a programmer its very complicated work for me.

boelle, 2020-06-21:
@TheCloudIdea still an issue?

astorun, 2020-06-21:
Yes its still not working as far as I know.

boelle, 2020-06-21:
have you tested it recently? was updated 13 hours ago

boelle, 2020-06-29:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

astorun, 2020-07-04:
Thanks for reopening this. I am gracefully waiting for humidity and temperature sensor support. :)