MarlinFirmware/Marlin #17558: [FR] Add compiler directive to disable PID Auto-Tuning in Configuration.h (to reduce program size)

jan-gerard, 2020-04-15:
### Description
Add compiler directive to disable PID Auto-Tuning in Configuration.h (e.g. PID_AUTOTUNE).
Now I can disable PIDTEMP or PIDTEMPBED, but not the PID **tuning feature** separately.
In order to reduce the firmware size, for example to enable SDSUPPORT and LEVELING, this could help. The PID-parameters can be tuned manually, or by using default values for a specific printer model.

### Feature Workflow
1. Disable a new directive #define PID_AUTOTUNE
2. Enable features that otherwise would not fit on a limited-memory size board.
3. Compile the firmware
4. Upload the firmware

#### Additional Information
For example for an Anet A6 printer.

jan-gerard, 2020-04-15:
I managed to reduce the binary file size by 3112 bytes. Now I can have my Anet A6 include:
- AUTO_BED_LEVELING_LINEAR
- LCD_BED_LEVELING
- SDSUPPORT
- LIN_ADVANCE
- BLTOUCH

sjasonsmith, 2020-09-11:
@jan-gerard if you have already implemented this for your own use, would you consider preparing a pull request to complete this yourself?