MarlinFirmware/Marlin #17559: Error compiling for ramps 1.4 (#error "Unknown MOTHERBOARD value set in Configuration.h")

crazytacowizerd, 2020-04-15:
Hi, I recently needed to reupload marlin 2 to my printer but I got an error compiling saying that there was an unknown motherboard value selected but there is one selected I copied it from the boards.h file. the board I am using the ramps 1.4, so my motherboard value looks like "#define BOARD_RAMPS_14_EFB            1020" and yet it still says that there isn't a correct value set. Can anyone please help me resolve this issue, please.
[image]
I have previously uploaded marlin 2 to my printer but I needed to do it again and I haven't changed anything it just stopped uploading.

ellensp, 2020-04-15:
You are in error. It should be 
#define MOTHERBOARD BOARD_RAMPS_14_EFB

And this is not a support forum. 
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted [on this page.](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question)) For best results getting help with configuration and troubleshooting, please use the following resources:

MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: [https://discord.gg/n5NJ59y](https://discord.gg/n5NJ59y)
After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

crazytacowizerd, 2020-04-15:
Sorry I didn't realize this wasn't a support forum but thank you for your help anyway :)

ellensp, 2020-04-15:
Please close this job, and I hope one of the actual support forums and live chats can assist with further queries.

closed 2020-04-17 by thinkyhead