MarlinFirmware/Marlin #17571: [BUG] Minimal extrude temp does not stop the print

Exalik, 2020-04-16:
### Bug Description
If for any reason the nozzle temp goes bellow the minimale extrude temp, the extrudeur stops but the printer continue to print.

### My Configurations

[Marlin 2.0.5.3 Config.zip](https://github.com/MarlinFirmware/Marlin/files/4486049/Marlin.2.0.5.3.Config.zip)

### Steps to Reproduce

1. Launch a print
2. After the first layer (bed layer), set the nozzle print temp at 0
(For me I think there's a bug in my gcode file because the nozzle temp target set to zero just after the bed layer temp)
3. Look the nozzle temp to decrease and goes bellow the minimale extrude temp
4. Look the extrudeur stops but the printer does not! (continue to print)

**Expected behavior:** The printer should pause or stop the print

**Actual behavior:** The printer continue to print

#### Additional Information

Geeetech A10M (GT2560 v3.1) with TMC2208 (Standalone mode)

AnHardt, 2020-04-16:
You should get "Cold Extrusion" warnings.
Up to now this is only a warning. Nothing bad should happen - except wasting a bit of time.

Exalik, 2020-04-16:
Hello AnHardt, thank's for your help.

Is it Tools / Cold Extrude Off in the menu please ?

thinkyhead, 2020-04-17:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

Exalik, 2020-04-18:
Hello,
I have made a Cold Extrude Off and, when the problem has occured another time, the extrudeur does not have been stopped.
So, if the nozzle temperature goes bellow the minimale extrude temperature, the extrusion can be stopped but the printer continue to print...

closed 2020-06-20 by boelle