MarlinFirmware/Marlin #17573: [BUG bf2.0] (LCD fanspeed issue + all_axes_homed return always false)

studiodyne, 2020-04-16:
Two bugs 👍 
All is in the movie included
[Vid](https://www.youtube.com/watch?v=2H3KPlqRC9Q
)
All_axes_homed() return always false :  
- Tool migration and many , need homing before starting ' in the video you can see the migration begins only if not homed' , and tool change too

LCD fan speed corrupted : Impossible to set it by LCD and all commands from lcd that need fanspeed is corrupted , because UI send a 0 value eternally

Help: I use UI for homing , UI for fanspeed , i'm sure UI lock anything that is sent by LCD
and surely the same here [#17570](https://github.com/MarlinFirmware/Marlin/issues/17570)

thinkyhead, 2020-04-17:
When you add logging to print out the current value of `axis_homed` what does it say?

studiodyne, 2020-04-17:
> 
> 
> When you add logging to print out the current value of `axis_homed` what does it say?
When i try to prime() , a function that only have ' all_axes_homed' , the prime doesn't work when homed , and works when not
When i've tried to migrate or tool change after LCD HOMING ' by custom user menu' , same problem
And same by SDPRINT

closed 2020-04-23 by studiodyne

thinkyhead, 2020-05-01:
I simply don't have time to become a clone of the expert (especially when this information is made extra hard to read). If you are an expert on the things you are writing about, send us a patch.