MarlinFirmware/Marlin #17576: [BUG] (short description) Teensy 3.5 no menu, no serial.

xsiriox, 2020-04-16:
### Bug Description

Hello, I just compiled a basic Marlin on my  teensy 3.5, but is not working.
I can compile it: in my display (Panelolu2) I can see the Marlin bootlogo, and then nothing.
I can use Serial Monitor in arduino to see what appens, and this is the code:

Marlin bugfix-2.0.x

echo: Last Updated: 2020-04-14 | Author: (none, default config)
echo:Compiled: Apr 16 2020
echo: Free Memory: 253232  PlannerBufferBytes: 1344
echo:Hardcoded Default Settings Loaded
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius

echo:; Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:; Steps per unit:
echo: M92 X80.00 Y80.00 Z4000.00 E500.00
echo:; Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z5.00 E25.00
echo:; Maximum Acceleration (units/s2):
echo:  M201 X3000.00 Y3000.00 Z100.00 E10000.00
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
echo:  M205 B20000.00 S0.00 T0.00 J0.01
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; Material heatup parameters:
echo:  M145 S0 H180 B70 F0
echo:  M145 S1 H240 B110 F0
echo:; PID settings:
echo:  M301 P22.20 I1.08 D114.00

If I send command, es M502, nothing append, it seems hanged.

I just see the versions of @teemuatlut: the version at https://github.com/teemuatlut/Marlin/tree/Teensy-HAL is working, new version for Marlin 2 at https://github.com/teemuatlut/Marlin/tree/Teensy-new can boot normally with display and it is not going to hang in Serial Monitor.

Is it possible to fix the current Marlin just to be able to start using the teensy 3.5/3.6?

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/4487621/Configs.zip)

### Steps to Reproduce

1. [Change the Configuration.h for motherboard and display]
2. [compiling and uploading]

**Expected behavior:** [What you expect to happen]

Marlin boots normally

**Actual behavior:** [What actually happens]

Marlin Hangs

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

boelle, 2020-06-20:
@xsiriox is the issue still there?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle