MarlinFirmware/Marlin #17581: Mesh Bed Leveling (MBL) without LCD not working

anjoze, 2020-04-17:
### Bug Description
I've enabled Mesh Bed Leveling (MBL).
I can't make it work without LCD because Z doesn't move to negative values.
All the videos and tutorials are with LCD but I want without LCD as described here:
https://marlinfw.org/docs/gcode/G029-mbl.html

**Mesh Bed Leveling from the host:**
1 Use G29 S0 to get the current status and mesh. If there’s an existing mesh, you can send M420 S1 to use it.
2 Use G29 S1 to move to the first point for Z adjustment.
3 Adjust Z so a piece of paper can just pass under the nozzle.
4 Use G29 S2 to save the Z value and move to the next point.
5 Repeat steps 3-4 until completed.

Can't move Z to negative values

### My Configurations

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4491916/Configuration.zip)

### Steps to Reproduce

I have a Prusa clone and my bed is taller on center than some of the corners.
The homing is on the center with a Z MIN ENDSTOP

STEPS:
1. G28 to home (center of bed)
2. G29 S1 to move to the first point for Z adjustment, or, use G29 S2 to save the Z value and move to the next point.

**Expected behavior:** [What you expect to happen]
If I need to adjust Z to a lower than 0  It should be possible

**Actual behavior:** [What actually happens]
Even if I disable the software endstops it will not go to negative values.

boelle, 2020-06-20:
@anjoze is the issue still there?

anjoze, 2020-06-21:
@boelle I think it is.
I can't confirm because since I opened the bug I did no updated, but I have the 2.0 firmware.

boelle, 2020-06-21:
you need to retest with https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

its updated daily

anjoze, 2020-06-21:
Ok. Maybe in 2 or 3 weeks I will be able to update it, than I'll give the feedback.
Thank you

anjoze, 2020-08-07:
Tested with 2.0.6
After testing and testing I think my problem is that I have the Z Endstop on bottom (z-min).
When the physical switch is activated it will not move down anymore.
Probably It it not supposed to work with this configuration and it is meant to work only with Z-Max switch.
Can you confirm this? Thanks you

boelle, 2020-08-08:
is it the same with [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) ?

anjoze, 2020-08-10:
> is it the same with [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) ?

Because I as not 100% sure I've downloaded [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) and try it again without success.
I've attached my config files.
My printer have a Z-Min endstop like this: https://i.pinimg.com/originals/71/08/f6/7108f6e3bc3a1a790549217642b4ba5e.jpg 
Is it supposed to go above 0 when the switch is triggered? Or just works with Z-Max endstop?

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5050435/Configuration.zip)

closed 2020-09-15 by github-actions[bot]