MarlinFirmware/Marlin #17582: [BUG] Power loss enabled but recovery menu does not show up

trustfm, 2020-04-17:
### Bug Description
At version 2.0.5 when i enable power loss the feature works (if an power loss happens then the resume menu appoears next time the machine starts up)
This does not happen at 2.0.5.3 (the resume menu does not appear after a power loss )

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

My confinguration files 
[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/4492851/Configurations.zip)

### Steps to Reproduce

1. [First Step]
start printin
2. [Second Step]
power loss happens
3. [3rd step]
when the machine starts there is no resume menu

thinkyhead, 2020-04-17:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

trustfm, 2020-04-17:
ok please give me a minute to test ... 
Thanks in advance ...

trustfm, 2020-04-17:
i am back i can confirm that the bug is fixed . So the 2.0.5.3 bugixed version works like the 2.0.5 

Thanks !

One small request , The first time that the power runout happens the POWER_LOSS_ZRAISE is triggered correctly and after that in the next boot the "resume menu" appears correctly. Now if the user is not present and another power runout happens the  LOSS_ZRAISE is triggered again incorrectly. If the  "resume menu" is present and no action is taken by the user this shoulld reappear on the next boot as many times as this happens without doing any LOSS_ZRAISE. 
Otherwise that procedure is correct only if the machine has a "No Volt Release Switch NVR"

closed 2020-06-20 by boelle