MarlinFirmware/Marlin #17587: [BUG] ABL Bilinear doesnt compensate Level

ils15, 2020-04-18:
Hello

Bilinear Leveling Grid:
      0      1      2      3      4
 0 -0.153 +0.015 -0.072 +0.125 +0.190
 1 -0.148 +0.035 -0.028 +0.197 +0.282
 2 -0.028 +0.080 -0.040 +0.137 +0.190
 3 +0.160 +0.185 -0.008 +0.115 +0.082
 4 +0.237 +0.280 +0.142 +0.292 +0.342

This my Level Grid and issue Is

ABL doesnt compesate Grid 

try several things and not happens

Z Offset still configurated (at -2.45)
i dont know what i can do more

ATmega2650 (Longer LK4 Pro)

Grogyan, 2020-04-18:
Try adding
M420 S1
To your start gcode.  It shouldn't be necessary to enable the current mesh.  Something to try

ils15, 2020-04-18:
I've already done it

Enviado do Outlook Mobile<https://aka.ms/blhgte>

thierryzoller, 2020-04-19:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/17399 and https://github.com/MarlinFirmware/Marlin/issues/17394 and https://github.com/MarlinFirmware/Marlin/issues/17386 andhttps://github.com/MarlinFirmware/Marlin/issues/17349 ?

polato94, 2020-04-30:
i had the same issue

change `define Z_PROBE_LOW_POINT -2`
to
`define Z_PROBE_LOW_POINT -10`
and try again

thierryzoller, 2020-05-01:
If that's the case then https://github.com/MarlinFirmware/Marlin/issues/14406 needs to be reopened. Another bug that was closed without solving the issue.

polato94, 2020-05-01:
Could be the same issue.

but after changing the Low-Point, my bed levling worked again. So i thought it will be a solution maybe.

Mils24, 2020-05-18:
I have just tried changing define Z_PROBE_LOW_POINT from -2 to -10 and can confirm it still does not make my bilinear mesh apply to my print.

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

thierryzoller, 2020-06-21:
In my book an issue is solved if code exists that fixes the issue OR where the issue was non existing. Why are we continuing to close  valid and unsolved reports?

boelle, 2020-06-21:
it was an dublicate of other issues and @ils15 seemed to have a fix for the issue

closed 2020-06-29 by boelle