MarlinFirmware/Marlin #17590: [FR] Preheat Constants

android444, 2020-04-18:
I analyzed the code and noticed such a thing.

```
case 0: // Preheat PLA
        #if defined(PREHEAT_1_TEMP_HOTEND) && defined(PREHEAT_1_TEMP_BED)
          e_temp = PREHEAT_1_TEMP_HOTEND;
          bed_temp = PREHEAT_1_TEMP_BED;
        #endif
        break;
      case 1: // Preheat ABS
        #if defined(PREHEAT_2_TEMP_HOTEND) && defined(PREHEAT_2_TEMP_BED)
          e_temp = PREHEAT_2_TEMP_HOTEND;
          bed_temp = PREHEAT_2_TEMP_BED;
        #endif
        break;
      case 2: // Preheat PET
        #if defined(PREHEAT_3_TEMP_HOTEND) && defined(PREHEAT_3_TEMP_BED)
          e_temp = PREHEAT_3_TEMP_HOTEND;
          bed_temp = PREHEAT_3_TEMP_BED;
        #endif
        break;
      case 3: // Preheat FLEX
        #if defined(PREHEAT_4_TEMP_HOTEND) && defined(PREHEAT_4_TEMP_BED)
          e_temp = PREHEAT_4_TEMP_HOTEND;
          bed_temp = PREHEAT_4_TEMP_BED;
        #endif

```
In <configurations.h> there is only PLA and ABS.
Can you add an additional preheat?

GMagician, 2020-04-18:
that's present under `#if ENABLED(DGUS_PREHEAT_UI)` not on standard lcd (it seems)

android444, 2020-04-18:
ok

thisiskeithb, 2020-04-18:
You can only add two right now for the main LCD code, but it would still be nice to increase the current limit of 2:

https://github.com/MarlinFirmware/Marlin/blob/8ad056013cd14907dc1a2aebdc0f4a31308606e2/Marlin/src/module/configuration_store.cpp#L815-L829

android444, 2020-04-18:
I want to increase it.
Only this requires a general rewriting of the code.
I want to choose any amount - minimum 1.

GMagician, 2020-04-18:
Need to increse also somewhere else. The reported code is only for eeprom save....

thisiskeithb, 2020-04-18:
> Need to increase also somewhere else. The reported code is only for eeprom save....

Updated my link 😉

sl1pkn07, 2020-04-19:
https://github.com/sl1pkn07/Marlin/commit/21dc00f2e54ad6270f1b3738760dc6a63d2aa7b2

I have aded a 3rd preheat (sorry for no separate in other commit. But is more easy for me if need rebase if upstream change the main code)

android444, 2020-04-19:
I want to do it so THINKYHEAD will accept it.

android444, 2020-04-19:
Whoever wrote this did not think about changes.
I don't really know how to change some things to make them compile.

thisiskeithb, 2021-09-24:
You can now have up to six preheat options, so just copy & paste one of the preheat constant blocks in `Configuration.h` and increase the `PREHEAT_N_` value for each new material.

closed 2021-09-24 by thisiskeithb