MarlinFirmware/Marlin #17594: error: 'X_HOME_BUMP_MM' was not declared in this scope

A-Caccese, 2020-04-18:
### Bug Description

Hi, I'm tring to compile Marlin 2.0.5.3 with Anet A8 configuration files, but I'm having this error:

```
Compiling .pio\build\sanguino1284p\src\src\module\scara.cpp.o
Marlin\src\module\motion.cpp:77:71: error: 'X_HOME_BUMP_MM' was not declared in this scope
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                       ^
Marlin\src\module\motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^
Marlin\src\module\motion.cpp:77:80: error: 'Y_HOME_BUMP_MM' was not declared in this scope
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                                ^
Marlin\src\module\motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^
Marlin\src\module\motion.cpp:77:89: error: 'Z_HOME_BUMP_MM' was not declared in this scope
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                                         ^
Marlin\src\module\motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^
*** [.pio\build\sanguino1284p\src\src\module\motion.cpp.o] Error 1
```

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4497074/Marlin.zip)

### Steps to Reproduce

1. Open Visual Studio Code
2. Open folder --> Marlin-2.0.x
3. Go to "Auto Build Marlin" --> Build

GMagician, 2020-04-18:
Try with bugfix before, but pay attention cnf is changed, now bump is for xyz (array) no more single for each axis

LukasOndracek, 2020-04-18:
same problem in Arduino

A-Caccese, 2020-04-18:
Bugfix was resolutive, thanks.

closed 2020-04-18 by A-Caccese

mattgb666, 2020-04-19:
Just hit the same issue in Arduino for Marlin 2.0.5.3 sketch\src\module\motion.cpp:77 (error below). Can you please elaborate on "try with bugfix before"? What would be the exact change I need to make to motion.cpp or other related files in the src folder?

Errors from Arduino:

`Arduino: 1.8.12 (Windows 10), Board: "Anet V1.0"

sketch\src\module\motion.cpp:77:71: error: 'X_HOME_BUMP_MM' was not declared in this scope

 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }

                                                                       ^

sketch\src\module\motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'

 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);

 ^~~~~~~~~~

sketch\src\module\motion.cpp:77:71: note: suggested alternative: 'HOMING_BUMP_MM'

 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }

                                                                       ^`

mattgb666, 2020-04-20:
Never mind. I found the 2.0.x bugfix branch and latest version there is now compiling! Cheers for all the hard work on this project.

jakebee09, 2020-04-22:
'Try with bug fix before' isn't sufficient enough info. I'm having the same problem are you able to clear it up a bit more on what to do? Thanks!

jakebee09, 2020-04-22:
> Try with bugfix before, but pay attention cnf is changed, now bump is for xyz (array) no more single for each axis

'Try with bug fix before' isn't sufficient enough info. I'm having the same problem are you able to clear it up a bit more on what to do? Thanks! Would really appreciate it :)

jakebee09, 2020-04-22:
> Never mind. I found the 2.0.x bugfix branch and latest version there is now compiling! Cheers for all the hard work on this project.

OR if you could direct me to your fix!

GMagician, 2020-04-22:
bugfix is new version, it should be tested before posting any new issue. That said, everytime a new version is released (bugfix may be always checked), configuration may change. In this case, a PR changed how  bump is defined.

now definition is:
#define HOMING_BUMP_MM      { 5, 5, 2 }       // (mm) Backoff from endstops after first bump

LittleDevil76, 2020-04-22:
Same problem as the string opener here, and yes i have tried Bugfix already.. 

(Trying on a original Ender 3)

GMagician, 2020-04-22:
@LittleDevil76 same error? motion.cpp line 77? Are you sure to have bugfix2? on such line in motion file there is a:
`/**`
sometime there is a mixing files between old and new version. Try to save your cnf, remove your marlin folder, re-download it and overwrite config with your files.
Then post you cnf to check them

bew1769, 2020-04-25:
I had both branches open in my workspace in VSCode.  Removed 2.0 branch from my workspace.  Compile Win

LittleDevil76, 2020-04-25:
@GMagician Yes tried what u mention, no change. But its only with the Ender 3 config.. A bit strange :/
I have a BIGTREETECH SKR mini E3 V1.2 32Bit on the way with the touch screen aswell, so im not that worried..
Tnx for the help tho..

roomme13, 2020-04-25:
I think they tried to merge bug fix to the 2.0.
You can either change configuration file to 2.0.5 or use the bugfix branch.

ellensp, 2020-04-25:
All you do is take the following lines int your Configuration_adv.h 
```
#define X_HOME_BUMP_MM 5
#define Y_HOME_BUMP_MM 5
#define Z_HOME_BUMP_MM 2
```
And change it into one new line

```
#define HOMING_BUMP_MM { 5, 5, 2 } // (mm) Backoff from endstops after first bump
```

Its no big deal.

edduda1, 2020-04-28:
/**
 * Homing Procedure
 * Homing (G28) does an indefinite move towards the endstops to establish
 * the position of the toolhead relative to the workspace.
 */

//#define SENSORLESS_BACKOFF_MM  { 2, 2 }     // (mm) Backoff from endstops before sensorless homing

#define HOMING_BUMP_MM      { 5, 5, 2 }       // (mm) Backoff from endstops after first bump
#define HOMING_BUMP_DIVISOR { 2, 2, 4 }       // Re-Bump Speed Divisor (Divides the Homing Feedrate)

//#define HOMING_BACKOFF_POST_MM { 2, 2, 2 }  // (mm) Backoff from endstops after homing
getting these errors:
Marlin\src\module\motion.cpp:77:71: error: 'X_HOME_BUMP_MM' was not declared in this scope
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                       ^
Marlin\src\module\motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^
Marlin\src\module\motion.cpp:77:80: error: 'Y_HOME_BUMP_MM' was not declared in this scope
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                                ^
Marlin\src\module\motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^
Marlin\src\module\motion.cpp:77:89: error: 'Z_HOME_BUMP_MM' was not declared in this scope
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                                         ^
Marlin\src\module\motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);

Does something need to change here?

#define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
motion.ccp
XYZ_CONSTS(float, base_min_pos,   MIN_POS);
XYZ_CONSTS(float, base_max_pos,   MAX_POS);
XYZ_CONSTS(float, base_home_pos,  HOME_POS);
XYZ_CONSTS(float, max_length,     MAX_LENGTH);
XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
XYZ_CONSTS(signed char, home_dir, HOME_DIR);

fikretduru, 2020-04-30:
rety X_HOME_BUMP_MM not work

SidShetye, 2020-05-23:
So what's the stable/production Marlin 2.x branch? 

I'm getting the same error on the release tagged `2.0.5.3` but this thread suggests using branch `bugfix-2.0.x`. But the readme there says ..

>Not for production use. Use with caution!

p3p, 2020-05-23:
I think this is caused by people downloading the config pack for the wrong release of Marlin, if you download Marlin 2.0.5.3 you need to make sure your when you download from https://github.com/MarlinFirmware/Configurations that you select the appropriate release branch (https://github.com/MarlinFirmware/Configurations/tree/release-2.0.5), if you just download the default one you will get the bugfix-2.0.x config pack.

SidShetye, 2020-05-23:
> I think this is caused by people downloading the config pack for the wrong release of Marlin, if you download Marlin 2.0.5.3 you need to make sure when you download from https://github.com/MarlinFirmware/Configurations that you select the appropriate release branch (https://github.com/MarlinFirmware/Configurations/tree/release-2.0.5), if you just download the default one you will get the bugfix-2.0.x config pack.

This is exactly right in the sense that the build compiles. In terms of testing/stability, are the tagged releases considered stable for 'production' usage? Granted 'production' is me and my kids playing with the Ender 3 but still we'd rather test less and tinker more :)

p3p, 2020-05-23:
The tagged releases are generally considered more stable than bugfix-2.0.x as that branch is the working development branch, unfortunately people don't tend to volunteer to be QA testers so other than the automated testing and user feedback they are really only production ready in the sense that they don't constantly get new updates.

cflouzat, 2020-05-26:
Hi
I'm far from a Marlin expert, but it appears to me the problem is not really in the code of Marlin, but the configuration examples we use and especially the *Configuration_adv.h* file.

From the discussion above, here is my humble turn around, which worked for me: 
* Open *Configuration_adv.h* with your favorite editor
```bash
vim Configuration_adv.h
```
* Find and replace the line
```c
#define HOMING_BUMP_MM      { 5, 5, 2 }       // (mm) Backoff from endstops after first bump
```
with
```c
#define X_HOME_BUMP_MM 5
#define Y_HOME_BUMP_MM 5
#define Z_HOME_BUMP_MM 2
#define HOMING_BUMP_MM      { X_HOME_BUMP_MM, Y_HOME_BUMP_MM, Z_HOME_BUMP_MM }       // (mm) Backoff from endstops after first bump
```
Doing things this way, my goal is to keep compatibility with today's implementation and prepare switch to next way to configure the BUMPs...

' hope it helps :)

Cheers
Fangorn_fr

gmar1274, 2020-05-28:
> All you do is take the following lines in your Configuration_adv.h
> 
> ```
> #define X_HOME_BUMP_MM 5
> #define Y_HOME_BUMP_MM 5
> #define Z_HOME_BUMP_MM 2
> ```
> 
> And change it into one new line
> 
> ```
> #define HOMING_BUMP_MM { 5, 5, 2 } // (mm) Backoff from endstops after first bump
> ```
> 
> Or the other way around if your trying to run a new config on an older marlin.
> Its no big deal.

> Hi
> I'm far from a Marlin expert, but it appears to me the problem is not really in the code of Marlin, but the configuration examples we use and especially the _Configuration_adv.h_ file.
> 
> From the discussion above, here is my humble turn around, which worked for me:
> 
> * Open _Configuration_adv.h_ with your favorite editor
> 
> ```shell
> vim Configuration_adv.h
> ```
> 
> * Find and replace the line
> 
> ```c
> #define HOMING_BUMP_MM      { 5, 5, 2 }       // (mm) Backoff from endstops after first bump
> ```
> 
> with
> 
> ```c
> #define X_HOME_BUMP_MM 5
> #define Y_HOME_BUMP_MM 5
> #define Z_HOME_BUMP_MM 2
> #define HOMING_BUMP_MM      { X_HOME_BUMP_MM, Y_HOME_BUMP_MM, Z_HOME_BUMP_MM }       // (mm) Backoff from endstops after first bump
> ```
> 
> Doing things this way, my goal is to keep compatibility with today's implementation and prepare switch to next way to configure the BUMPs...
> 
> ' hope it helps :)
> 
> Cheers
> Fangorn

Confirming this is the solution!!! Thank you @ellensp and @fangornfr 😃

tophat17, 2020-06-14:
I had a library issue. After uninstalling the library the IDE was complaining at, it worked!